ocerman/zenpower #4: Ryzen 3000 series support?

zaggynl, 2019-07-19:
Had a go at this with my 3900X but only power is shown: http://i.imgur.com/mcGUFmx.png

ocerman, 2019-07-19:
Thank you for your report.

I have made an update that should fix that. Can you please try it again?

zaggynl, 2019-07-20:
Thanks for the reply, no joy with updated version but happy to keep trying.

nickbuerger, 2019-07-25:
I added a pull request for Ryzen 3000 support.

zaggynl, 2019-07-25:
Thanks!

I had success with the following: https://old.reddit.com/r/Amd/comments/cer8s9/3900xgigabyte_x570_elite_working_well_on_linux/euj0wah/

`sudo modprobe it87 ignore_resource_conflict=1 force_id=0x8622`

Make permanent:

In /etc/modules-load.d/sensors.conf

it87

then in /etc/modprobe.d/sensors.conf

options it87 ignore_resource_conflict=1 force_id=0x622

closed 2019-07-25 by ocerman

zaggynl, 2019-07-26:
xsensors/zenmonitor now shows more data but appears incorrect: 

zenpower:
http://i.imgur.com/GLaU8Gv.png

it87:
http://i.imgur.com/YesVJzu.png

nickbuerger, 2019-07-26:
You need a kernel with the following patch for Ryzen 3000:
https://patchwork.kernel.org/patch/11043277/
Unfortunately it's not included yet in kernel 5.2 or 5.3.

I built my own kernel with this patch and the sensor data is shown correctly.
Only Core and SoC voltage seems to be mixed up.

ocerman, 2019-07-26:
Unfortunatelly, I do not own Ryzen 3000 CPU, so I cannot test this. 

> xsensors/zenmonitor now shows more data but appears incorrect
@nickbuerger is right, that patch seems to be needed. If it still does not work, let me know, I would reopen this issue.

>  Only Core and SoC voltage seems to be mixed up.
@nickbuerger  If something does not look right, let me know, I can reopen this issue, or you can open another one.

ridobe, 2019-07-27:
Ryzen 3900x:
The patch works great.  I think @nickbuerger is correct, the voltages seem switched.  Thanks for your hard work everyone.  This is/was sorely needed.