ocerman/zenpower #7: SVI2_Core & SVI2_SoC mixed up on ASRock X470 Taichi

RichGuk, 2019-08-10:
My Core and SoC seem to be the wrong way around; this is on a 3700x and ASRock X470 Taichi. It was correct when the CPU was in my old Asus X370 Prime board but on ASRock X470 Taichi it appears swapped. I have re-run sensors-detect etc. Unsure if I simply need to just rename some labels? If that's the case close the issue.

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.02 V  
SVI2_SoC:     +1.45 V  
Tdie:         +33.5°C  (high = +70.0°C)
Tctl:         +33.5°C  
SVI2_P_Core:  26.63 W  
SVI2_P_SoC:    6.28 W  
SVI2_C_Core: +25.98 A  
SVI2_C_SoC:   +4.33 A 
```

Shots of HWINFO:

[image]
[image]

and attached the debug info from it.
[HWiNFO64.DBG.zip](https://github.com/ocerman/zenpower/files/3488678/HWiNFO64.DBG.zip)

ocerman, 2019-08-10:
Hello, thanks for your report.
This is indeed a bug. Seems like some motherboards are reporting SoC and Core values in swapped addresses. Unfortunately AMD did not published any documentation for this so I do not how to detect which address is correct. Also Amps values do not seems to be accurate.

As a workaround you can actually rename labels in lm-sensors. You can do it by creating file `/etc/sensors.d/zenpower` with contents like this (this will swap SoC and Core voltage):
```
chip "zenpower-pci-*"
    label in1 "SVI2 SoC"
    label in2 "SVI2 Core"
```

wedens, 2019-08-12:
To add another data point: I have the same problem on MSI B450 Gaming Pro Carbon AC with Ryzen 3600 cpu.

ocerman, 2019-09-29:
I have another system with swapped values: ASUS ROG Strix X570-E Gaming with Ryzen 7 3700x

ocerman, 2019-10-06:
Hello @wedens, @RichGuk.
I have released the new version, can you try it? Seems like zen2 ryzen CPUs are reporting soc/core values on swapped addresses so the new version is expeting that.

Also @wedens do you still have that X370 board? Can you post either HWinfo or zenpower debug data (#12) ?

wedens, 2019-10-07:
> Also @wedens do you still have that X370 board? Can you post either HWinfo debug log or zenpower debug data (#12) ?

I don't have X370 board. I have B450 board. I think @RichGuk had X370.

I'll try the new version with my board soon and report back.

wedens, 2019-10-07:
I can confirm that with the new version results look like what I'd expect.

ocerman, 2019-10-12:
No, there was no temperature-related changes in last version

birdie-github, 2019-11-09:
This issue could be closed I guess.

closed 2019-11-11 by ocerman