ocerman/zenpower #8: Wrong info shows on Ryzen 3000

whatdamath, 2019-08-31:
First, thank you for creating this. Looking forward to new updates

Now for the issue:
I'm using Ryzen 3900x on top of Ubuntu 19.04 (Pop OS) and the numbers the sensor is showing are not correct.

This is what I'm getting from "sensors" command

```
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)
temp2:        +16.8°C  (crit = +20.8°C)

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    -0.04 V  
SVI2_SoC:     -0.04 V  
Tdie:        +206.9°C  (high = +70.0°C)
Tctl:        +206.9°C  
SVI2_P_Core: -11.39 W  
SVI2_P_SoC:   -3.96 W  
SVI2_C_Core: +265.00 A  
SVI2_C_SoC:  +92.00 A 
```

ridobe, 2019-09-01:
Did you apply the patch?

B2Corner, 2019-09-03:
+1. Have the exact same readings on a 3900X system with kernel 5.0.0-25-generic (which as far as I understand should have the Ryzen 3000 patches out-of-the-box).

ridobe, 2019-09-03:
Did you verify that the patch from the README has been applied? If not, you'll have to apply it. 5.3.0-rc6 doesn't contain it so 5.3 won't have it.

whatdamath, 2019-09-03:
If you at talking about the kernel patch - no because there is no easy way to do this on pop os - Ukuu relies on Grub and pop os is UEFI based. I had some success with k12temp though so I'm surprised it works there and not on zenpower. 
k12temp update does show more or less correct temperatures, so it definitely doesn't depend on the patch mentioned
The current kernel version for pop os is
Linux 5.0.0-27-generic x86_64

maded2, 2019-09-14:
have the same issue on my 3900x.  here's the output from zenmonitor

[image]

ocerman, 2019-09-29:
The patch is really required. The current kernel (5.3) still do not have that patch applied. However the next version (5.4) should work without any additional patches.

ridobe, 2019-10-04:
FYI, the patch has been included in 5.4rc1.

wedens, 2019-10-07:
5.3.4 also includes the patch

ridobe, 2019-10-07:
@wedens which distro?

wedens, 2019-10-07:
@ridobe Arch Linux. but it doesn't matter. it's upstream.

ocerman, 2019-10-07:
@wedens yes, you are right.
Thanks, I will update the readme file.

closed 2019-10-12 by ocerman

ocerman, 2019-11-04:
The new version of zenpower includes a fallback method for reading data from SMN, so when the linux kernel do not have support for your CPU a fallback method will be used instead of the kernel function.

The fallback method however may not work with some CPUs so having up-to-date kernel is still prefered. It works fine with my 3700x and stock Ubuntu 19.10 (without any kernel patches).