ocerman/zenpower #9: Investigating Accuracy of Vcore: Comparison with it87

Myrddin-Wyllt, 2019-09-04:
### [System Specifications](https://gist.github.com/Myrddin-Wyllt/4cc23dfec5512efc59b4d93336a6d36b)

I've been trying to gauge accuracy of Vcore readings compared to it87, & I've logged my results of Vcore at idle versus load with both. Simulated load was done using `mprime -t`. My system is set to 1.255V Vcore in the BIOS. 

### Here are my findings:
<pre>Idle           Load      Measure
--------------------------------
1.257V       |  1.25V   |  zenpower
1.243-1.254V |  1.243V  |  it87</pre>

I've been wondering now which method is more accurate. I understand it87 uses a variety of I/O chips which vary depending on motherboard manufacturer. After perusing your code, I noticed no reliance on such dodgy I/O chips. In your opinion, which do you think is more accurate & how would one devise which method of measurement is the most accurate. it87 seems dead in terms of development.

jiblime, 2019-11-18:
You may be interested in the discontinued external it87 kernel module. It is *far* ahead of even mainline's drivers/hwmon/it87.c.

Here is a read-only mirror: https://github.com/herocodemaster/it87

I was able to enable both the 'custom' it87 as well as zenpower's module and they are very close in temps. A bonus was that I got additional sensors detected for my ITE Super I/O chips it8686 and it8792. What I am confused is that I can't tell what is thermal throttling my PBO since the VRM temps on Windows with hwinfo are substantially lower than my CPU, as is the same on Linux. My motherboard is lambasted for its poor VRM cooling but they stay within 40-60C.

From my point of view, Guenter Roeck's version linked above would give proper ITE readings. Otherwise I would default to zenpower for readings. 

Question: what are you using to parse voltages from different modules as the same time? I am only able to get temperature. 

Mainline kernel 5.4.0-rc8, custom it87, with HEAD branch zenpower module:

```
it8686 1              40C
it8686 2              44C
it8686 3              58C
it8686 4              43C
it8686 5              44C
it8686 6              47C
it8792 1              30C
it8792 2              34C
it8792 3              36C
Tdie                  58C
Tctl                  58C
Tccd1                 63C
iwlwifi 1             51C
edge                  49C
it8686 1            2647R
it8686 2               0R
it8686 3               0R
it8686 4            1049R
it8686 5               0R
it8792 1               0R
it8792 2               0R
it8792 3               0R
amdgpu 1             782R
```

Myrddin-Wyllt, 2020-01-02:
I had noticed the groeck it87 out of tree module & that's what I've been comparing zenpower to. My testing was conducted sequentially; I first removed it87 before switching to zenpower. I never attempted to read voltages from both simultaneously; but theoretically that should be possible as it87 uses IO chips whereas zenpower used SVI2 on the CPU.

Zenpower works more reliably than groeck's it87 for most motherboards & is what I'd recommend moving forward. I'd still like to know which is the most accurate method of measure; something that likely won't get an answer.

closed 2023-12-19 by Myrddin-Wyllt