ocerman/zenpower #11: Speaking of SVI2_P_Core

birdie-github, 2019-09-28:
What does this sensor mean exactly?

It surely doesn't report the correct power consumption because it remains fairly constant (~31W) despite any load, while e.g.

`turbostat --quiet --num_iterations 1 --interval 1 --show PkgWatt --Summary`

reports the correct values (from 15W to 91W for my Ryzen 7 3700X CPU).

ocerman, 2019-09-29:
Core power (SVI2_P_Core) = Core current (SVI2_C_Core) * Core voltage (SVI2_Core)

However the Core current is not accurate on some systems so reported wattage/amperage may not be accurate (I have updated the readme stating about this issue).

ocerman, 2019-09-29:
@birdie-github can you please try to update zenpower to new version and post debug data to #12 ?

birdie-github, 2019-09-29:
Ryzen 7 3700X on a X570 motherboard.
```
0005a008 = 00000002
0005a00c = 014a001b
0005a010 = 01660003
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000a90
00059958 = 00000000
0005995c = 00000000
```

ocerman, 2019-10-06:
Hello, I have released the new version, can you try it? The old version had soc/core values swapped and the formulas for wattage/amperage were not accurate. This should be fixed.

birdie-github, 2019-10-20:
I've compiled the master. The values are still not correct (relatively idle):

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.91 V
SVI2_SoC:     +1.09 V
Tdie:         +38.4°C  (high = +70.0°C)
Tctl:         +38.4°C
SVI2_P_Core:   1.80 W
SVI2_P_SoC:    8.65 W
SVI2_C_Core:  +1.98 A
SVI2_C_SoC:   +7.95 A
```

turbostat shows around 25W.

Under load

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.33 V
SVI2_SoC:     +1.09 V
Tdie:         +55.6°C  (high = +70.0°C)
Tctl:         +55.6°C
SVI2_P_Core:  15.87 W
SVI2_P_SoC:    9.28 W
SVI2_C_Core: +11.86 A
SVI2_C_SoC:   +8.53 A
```

turbostat shows around 65W

ocerman, 2019-11-08:
So I have done some testing today with my 3700x. With `7z b` SVI2 Core power is ~42W and turbostat shows ~90W.

Because AMD did not publish documentation about SVI2 stuff, my goal is to show same values as [HWiNFO](https://www.hwinfo.com/). And on my system it seems all right (screenshots below).

According to HWiNFO developer ([source](https://www.hwinfo.com/forum/threads/cpu-package-cpu-core-power.5473/#post-20285)) there is a difference between SVI2 (SVI2_P_Core) and CPU Package Power (smu/rapl/turbostat):

> AMD Zen CPUs have multiple voltage rails: VDDCR_CPU and VDDCR_SOC
CPU Core Power (SVI2 TFN) is measured via SVI2 telemetry and it reflects only the core (VDDCR_CPU) rail. SoC Power reflects the VDDCR_SOC rail. So VDDCR_CPU + VDDCR_SOC + (additional rest-of-chip power that can't be measured) should equal the total CPU Package Power.

> I suggest to watch the CPU Package Power (SMU) value only as this is the total telemetry provided internally by the CPU.

[image]

[image]

@birdie-github  16W under load seems wrong. The debug data that you have [posted](https://github.com/ocerman/zenpower/issues/11#issuecomment-536336212) were under load?

birdie-github, 2019-11-09:
After compiling the master:

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.29 V  
SVI2_SoC:     +1.09 V  
Tdie:         +54.5°C  (high = +70.0°C)
Tctl:         +54.5°C  
Tccd1:        +58.8°C  
SVI2_P_Core:  35.64 W  
SVI2_P_SoC:    9.61 W  
SVI2_C_Core: +27.01 A  
SVI2_C_SoC:   +8.83 A  
```

ocerman, 2019-11-09:
Yes, it is possible to export cpu package power to the hwmon kernel driver. I will try to do it soon, however I won't put it into this kernel module and instead I will make a new one.

closed 2019-11-09 by birdie-github