ocerman/zenpower #12: [Help needed] Please provide debug data from zenpower

ocerman, 2019-09-29:
Hello everyone,
It would be very helpful for me for further development of Zenpower if you can share debug data from zenpower. Zenpower debug data contains raw data from SMN.

If you have latest zenpower driver installed, you can get debug data from file `/sys/class/hwmon/hwmon[num]/debug_data` or via this [bash script](https://gist.github.com/ocerman/6566b00072c8af96a109f65075f2d8b7)

Please also state your CPU, Motherboard and Linux kernel version. Thank you.

ocerman, 2019-09-29:
ASUS ROG Strix X570-E Gaing
Ryzen 7 3700X
```
0005a008 = 00000002
0005a00c = 014a0023
0005a010 = 01630002
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000ad6
00059958 = 00000000
0005995c = 00000000
```

Kattus, 2019-09-29:
ASUS PRIME B350M-A
Ryzen 5 1600X
Linux 5.3.1
```
0005a008 = 00000002
0005a00c = 0136003b
0005a010 = 017f0014
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```

terroreek, 2019-09-29:
Asus ROG Strix X399-E Gaming
Threadripper 2920x
Linux 5.2.17

```
0005a008 = 00000002  
0005a00c = 0158003a    
0005a010 = 01f70000  
000598bc = 0fff00ff  
0005994c = 00000000  
00059954 = 00000b7a  
00059958 = 00000000  
0005995c = 00000000
```
and 
```
0005a008 = 00000002
0005a00c = 0119000f
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```

NoXPhasma, 2019-09-29:
Ryzen 7 1700
Asus Prime X370-Pro
Kernel 5.2.0-17.2-liquorix-amd64

```
0005a008 = 00000002
0005a00c = 0121000a
0005a010 = 015a001d
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```

Values seem to be correct.

birdie-github, 2019-09-29:
Ryzen 7 3700X on a X570 motherboard.
```
0005a008 = 00000002
0005a00c = 014a001b
0005a010 = 01660003
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000a90
00059958 = 00000000
0005995c = 00000000
```

Linux 5.2 with the two patches which allow the k10temp driver to recognize my CPU.

jrugia, 2019-09-29:
CPU - Ryzen 1700X
Motherboard - Asus PRIME B350M-A
Kernel 5.3.1

> 0005a008 = 00000002
> 0005a00c = 0139000c
> 0005a010 = 014a001a
> 000598bc = 0fff00ff
> 0005994c = 00000000
> 00059954 = 00000000
> 00059958 = 00000000
> 0005995c = 00000000

Output when idle (more or less):
> zenpower-pci-00c3
> Adapter: PCI adapter
> SVI2_Core:    +1.20 V  
> SVI2_SoC:     +1.09 V  
> Tdie:         +40.2°C  (high = +70.0°C)
> Tctl:         +60.2°C  
> SVI2_P_Core:  25.09 W  
> SVI2_P_SoC:   10.60 W  
> SVI2_C_Core: +18.70 A  
> SVI2_C_SoC:  +10.10 A 

During Prime95 Small FFTs (AVX enabled):
> zenpower-pci-00c3
> Adapter: PCI adapter
> SVI2_Core:    +1.09 V  
> SVI2_SoC:     +1.09 V  
> Tdie:         +63.4°C  (high = +70.0°C)
> Tctl:         +83.4°C  
> SVI2_P_Core:  91.58 W  
> SVI2_P_SoC:   11.84 W  
> SVI2_C_Core: +84.18 A  
> SVI2_C_SoC:  +10.10 A

crt0mega, 2019-10-03:
Ryzen R7 3700X
Asus Prime X370-PRO
5.2.0-17.2-liquorix-amd64

0005a008 = 00000002
0005a00c = 014a0023
0005a010 = 010a0009
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000ae0
00059958 = 00000000
0005995c = 00000000

wedens, 2019-10-07:
Ryzen 3600
MSI B450 Gaming Pro Carbon AC
Linux 5.3.4

0005a008 = 00000002
0005a00c = 014b001f
0005a010 = 011a0014
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000b40
00059958 = 00000000
0005995c = 00000000

Core/SOC values were swapped before the last version

sambolinux, 2019-10-14:
CPU: RYZEN 2700X
MB:ASUS CROSSHAIR VII HERO (WIFI) 
KERNEL: 5.3.6-050306
DEBUG_DATA:
0005a008 = 00000002
0005a00c = 01760006
0005a010 = 01750013
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000

ZENPOWER DATA:
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.81 V  
SVI2_SoC:     +0.82 V  
Tdie:         +30.8°C  (high = +70.0°C)
Tctl:         +40.8°C  
SVI2_P_Core:   4.22 W  
SVI2_P_SoC:    5.61 W  
SVI2_C_Core:  +5.20 A  
SVI2_C_SoC:   +6.85 A

jbaiter, 2019-10-16:
CPU: Ryzen 3900X
MB: Giogabyte X570 Aorus Ultra
Kernel: 5.2.17

```
0005a008 = 80000005
0005a00c = 80000005
0005a010 = 80000005
000598bc = 80000005
0005994c = 80000005
00059954 = 80000005
00059958 = 80000005
0005995c = 80000005
```

Doesn't seem to be working :-/

jjrams, 2019-10-19:
Ryzen 3900X
MSI X570-A Pro
Linux 5.3.6-arch1-1-ARCH

```
% ./zp_read_debug.sh
0005a008 = 00000002
0005a00c = 014b003a
0005a010 = 0127007a
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000c1a
00059958 = 00000bac
0005995c = 00000000
```

```
% sensors zenpower-pci-00c3
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.31 V
SVI2_SoC:     +1.08 V
Tdie:         +81.8°C  (high = +70.0°C)
Tctl:         +81.8°C
SVI2_P_Core: 105.91 W
SVI2_P_SoC:   18.47 W
SVI2_C_Core: +81.03 A
SVI2_C_SoC:  +17.07 A
```

lizardb0y, 2019-10-22:
MB: Gigabyte AB350N-Gaming-Wifi
CPU: Ryzen 5 2400G
Kernel Linux 5.0.0-32-generic #34-Ubuntu

0005a008 = 00000002
0005a00c = 016d0001
0005a010 = 0139000a
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000

Output from zenpower looks good. Under load (stress-ng --cpu=8):

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.34 V  
SVI2_SoC:     +1.19 V  
Tdie:         +64.1°C  (high = +70.0°C)
Tctl:         +64.1°C  
SVI2_P_Core:  41.71 W  
SVI2_P_SoC:    4.31 W  
SVI2_C_Core: +30.14 A  
SVI2_C_SoC:   +3.97 A

lucasxteixeira, 2019-11-02:
Gygabyte X570 Aorus Elite
Ryzen 9 3900X
Linux 5.3.8 (xanmod)

Debug data:
0005a008 = 00000002
0005a00c = 014c0028
0005a010 = 016d0009
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000ac4
00059958 = 00000acc
0005995c = 00000000

The output from zenpower for Core looks correct, the output for SoC is static (never changes):
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.31 V  
SVI2_SoC:     +1.07 V  
Tdie:         +48.9°C  (high = +70.0°C)
Tctl:         +48.9°C  
SVI2_P_Core:  11.19 W  
SVI2_P_SoC:   12.65 W  
SVI2_C_Core:  +8.56 A  
SVI2_C_SoC:  +11.77 A

birdie-github, 2019-11-07:
@lucasxteixeira 
> The output from zenpower for Core looks correct, the output for SoC is static (never changes):
> `SVI2_P_Core: 11.19 W`

Could you post your results under load as well? `7z b` will suffice.

lucasxteixeira, 2019-11-08:
> `7z b`
There you go:

```
Adapter: PCI adapter
SVI2_Core:    +1.23 V  
SVI2_SoC:     +1.07 V  
Tdie:         +66.8°C  (high = +70.0°C)
Tctl:         +66.8°C  
SVI2_P_Core:  56.82 W  
SVI2_P_SoC:   12.65 W  
SVI2_C_Core: +46.12 A  
SVI2_C_SoC:  +11.77 A
```

birdie-github, 2019-11-08:
@lucasxteixeira 

This doesn't look right. The Ryzen 9 3900X consumes over 115W under load and your values are really really low. Unless you somehow limited your CPU performance they are outright wrong. My 3700X consumes around 90W for this test.

Please try running [this script](https://github.com/birdie-github/useful-scripts/blob/master/watts_cpu) after launching `7z b`.

ocerman, 2019-11-08:
@birdie-github @lucasxteixeira let's move discussion about power to #11

cyboerg42, 2019-11-11:
AMD EPYC 7351P

```
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01610051
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000a7a
00059958 = 00000a70
0005995c = 00000a66
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 0140000a
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```

sensors output :

```
zenpower-pci-00d3
Adapter: PCI adapter
SVI2_Core:    +1.55 V  
SVI2_SoC:     +1.55 V  
Tdie:         +29.6°C  (high = +70.0°C)
Tctl:         +29.6°C  
SVI2_P_Core:   0.00 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core:  +0.00 A  
SVI2_C_SoC:   +0.00 A  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.94 V  
SVI2_SoC:     +0.01 V  
Tdie:         +30.5°C  (high = +70.0°C)
Tctl:         +30.5°C  
SVI2_P_Core:  76.52 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core: +81.06 A  
SVI2_C_SoC:   +0.00 A  

zenpower-pci-00db
Adapter: PCI adapter
SVI2_Core:    +1.55 V  
SVI2_SoC:     +1.55 V  
Tdie:         +27.9°C  (high = +70.0°C)
Tctl:         +27.9°C  
SVI2_P_Core:   0.00 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core:  +0.00 A  
SVI2_C_SoC:   +0.00 A  

zenpower-pci-00cb
Adapter: PCI adapter
SVI2_Core:    +1.15 V  
SVI2_SoC:     +0.01 V  
Tdie:         +30.4°C  (high = +70.0°C)
Tctl:         +30.4°C  
SVI2_P_Core:  13.15 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core: +13.51 A  
SVI2_C_SoC:   +0.00 A  
```

core voltage for p0 state is fixed to +1.15 V, so that's alright. - the iLO is also reporting the same avg cpu temp - so that should be fine too.

lots0logs, 2019-11-12:
Distro: Arch
Kernel: 5.3.10
CPU: Ryzen 7 3900x

```
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 014b002c
0005a010 = 010b0014
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000aee
00059958 = 00000ada
0005995c = 00000000
```
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.39 V
SVI2_SoC:      1.08 V
Tdie:         +70.6°C  (high = +70.0°C)
Tctl:         +70.6°C
Tccd1:        +72.0°C
Tccd2:        +42.8°C
SVI2_P_Core:  24.90 W
SVI2_P_SoC:   13.37 W
SVI2_C_Core:  17.79 A
SVI2_C_SoC:   12.36 A
```

Myrddin-Wyllt, 2019-11-13:
#### Motherboard: Biostar X370GT7
#### CPU: Ryzen 7 1700
#### Kernel: 5.3.7-arch1-2-fsync

### Debug Data:
<pre>kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 012e0004
0005a010 = 0149000a
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000</pre>

### sensors Data:
<pre>    IDLE:
=======================

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.26 V  
SVI2_SoC:      1.09 V  
Tdie:         +26.8°C  (high = +70.0°C)
Tctl:         +26.8°C  
SVI2_P_Core:  11.81 W  
SVI2_P_SoC:    3.55 W  
SVI2_C_Core:  11.43 A  
SVI2_C_SoC:    3.61 A</pre>
<pre>    LOAD (7z b):
=======================

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.26 V  
SVI2_SoC:      1.09 V  
Tdie:         +49.8°C  (high = +70.0°C)
Tctl:         +49.8°C  
SVI2_P_Core:  77.07 W  
SVI2_P_SoC:    4.32 W  
SVI2_C_Core:  61.31 A  
SVI2_C_SoC:    3.97 A </pre>

mbana, 2019-11-15:
*NB:* You could made it easier to run this script.

```sh
$ curl -s https://gist.githubusercontent.com/ocerman/6566b00072c8af96a109f65075f2d8b7/raw/4a015fe1d2b6eca95e38ce0400a023de57d901ab/zp_read_debug.sh | bash
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01480030
0005a010 = 01600001
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000a8c
00059958 = 00000a8e
0005995c = 00000000
```

```sh
$ sudo lshw
mbana-pc1                   
    description: Desktop Computer
    product: MS-7C36 (To be filled by O.E.M.)
    vendor: Micro-Star International Co., Ltd.
    version: 1.0
    serial: To be filled by O.E.M.
    width: 64 bits
    capabilities: smbios-2.8 dmi-2.8 smp vsyscall32
    configuration: boot=normal chassis=desktop family=To be filled by O.E.M. sku=To be filled by O.E.M. uuid=ACE4BF3A-C485-16BA-ABC4-00D861A3342D
  *-core
       description: Motherboard
       product: PRESTIGE X570 CREATION (MS-7C36)
       vendor: Micro-Star International Co., Ltd.
       physical id: 0
       version: 1.0
       serial: J816282762
       slot: To be filled by O.E.M.
     *-firmware
          description: BIOS
          vendor: American Megatrends Inc.
          physical id: 0
          version: 1.61
          date: 11/06/2019
          size: 64KiB
          capacity: 15MiB
          capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb biosbootspecification uefi
     *-memory
          description: System Memory
          physical id: f
          slot: System board or motherboard
          size: 64GiB
        *-bank:0
             description: Project-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719)Project-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719) 3000 MHz (0.3 ns) [empty]
             product: Unknown
             vendor: Unknown
             physical id: 0
             serial: Unknown
             slot: DIMM 0
             clock: 3GHz (0.3ns)
        *-bank:1
             description: DIMM DDR4 Synchronous Unbuffered (Unregistered) 3000 MHz (0.3 ns)
             product: CMK32GX4M1D3000C16
             vendor: Unknown
             physical id: 1
             serial: 00000000
             slot: DIMM 1
             size: 32GiB
             width: 64 bits
             clock: 3GHz (0.3ns)
        *-bank:2
             description: Project-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719)Project-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719) 3000 MHz (0.3 ns) [empty]
             product: Unknown
             vendor: Unknown
[... 1083 lines cut ...]
                logical name: /home
                version: 1.0
                serial: e67e1146-e715-475b-8f6c-cb1dc3c4a7d3
                size: 119GiB
                capabilities: journaled extended_attributes large_files huge_files dir_nlink recover 64bit extents ext4 ext2 initialized
                configuration: created=2019-11-05 16:45:35 filesystem=ext4 lastmountpoint=/home modified=2019-11-15 15:21:16 mount.fstype=ext4 mount.options=rw,relatime mounted=2019-11-15 15:21:16 state=mounted
           *-volume:6
                description: EXT4 volume
                vendor: Linux
                physical id: 7
                bus info: scsi@1:0.0.0,7
                logical name: /dev/sdb7
                version: 1.0
                serial: ce006e86-c7eb-47f8-b85d-95ef75249e0c
                size: 95GiB
                capabilities: journaled extended_attributes large_files huge_files dir_nlink recover 64bit extents ext4 ext2 initialized
                configuration: created=2019-11-05 16:11:58 filesystem=ext4 lastmountpoint=/home modified=2019-11-05 18:35:48 mounted=2019-11-05 18:35:48 state=clean
  *-network
       description: Ethernet interface
       physical id: 1
       logical name: docker0
       serial: 02:42:0b:33:69:cc
       capabilities: ethernet physical
       configuration: broadcast=yes driver=bridge driverversion=2.3 firmware=N/A ip=172.17.0.1 link=no multicast=yes
```

cyboerg42, 2019-11-15:
```curl -s https://gist.githubusercontent.com/ocerman/6566b00072c8af96a109f65075f2d8b7/raw/4a015fe1d2b6eca95e38ce0400a023de57d901ab/zp_read_debug.sh | bash```

i can not recommend this, because it's kinda bad practice... :/

be sure to know, what you run. :)

jiblime, 2019-11-18:
> `curl -s https://gist.githubusercontent.com/ocerman/6566b00072c8af96a109f65075f2d8b7/raw/4a015fe1d2b6eca95e38ce0400a023de57d901ab/zp_read_debug.sh | bash`
> 
> i can not recommend this, because it's kinda bad practice... :/
> 
> be sure to know, what you run. :)

"Safer" way, compat with bash and zsh: This prints out the contents of a random shell script to let you know what you're going to be doing.

`
zp_dbg=$(curl -s https://gist.githubusercontent.com/ocerman/6566b00072c8af96a109f65075f2d8b7/raw/4a015fe1d2b6eca95e38ce0400a023de57d901ab/zp_read_debug.sh) ; echo "${zp_dbg}" ; SHELL=/bin/bash printf '\nIf you believe this script to be safe, press any key to continue. Other wise, SIGINT or CTRL+C.\n' ; read ret && SHELL=/bin/bash sh -c ${zp_dbg}
`

Chlorophytus, 2019-12-19:
Threadripper 3960X with a Gigabyte Aorus Pro Wi-Fi and DDR4-3200.
```
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01f70000
0005a010 = 01710029
000598bc = 0fff0fff
0005994c = 0fff0fff
00059954 = 00000a8c
00059958 = 00000000
0005995c = 00000a8c
```

hydrian, 2019-12-27:
ASRock B450 Gaming K4 Firmware P3.40
Kernel: 5.0.0-37-generic #40~18.04.1-Ubuntu SMP 
AMD Ryzen 5 2600X 

$ bash Downloads/zp_read_debug.sh 
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01560017
0005a010 = 014a0020
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000

haromu, 2019-12-27:
Gigabyte B450M S2H
Ryzen 5 3400G
Kernel 5.4.6-arch3-1

```
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 016c0007
0005a010 = 014c0013
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```

notzed, 2020-01-06:
linux 5.4.7
ryzen 3900x
gigabyte x570I aorus pro wifi

kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 0150002a
0005a010 = 015f0003
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000a9e
00059958 = 00000aae
0005995c = 00000000

pshemk, 2020-01-15:
linux 5.3.13
ryzen 3900x
asus x470 pro

kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 014b0030
0005a010 = 0133005e
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000c54
00059958 = 00000c52
0005995c = 00000000

ShadowTeolog, 2020-01-16:
ubuntu 18.04 5.0.0-37-generic
AMD® Ryzen 5 3550h with radeon vega mobile gfx × 8
ASUS FX505DY

kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 017e0002
0005a010 = 01750014
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000

JustMaximumPower, 2020-01-16:
Ubuntu 19.04 5.5.0-050500rc6-lowlatency
AMD Ryzen 7 3700X
MSI B450M PRO-VDH MAX

kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 014c001f
0005a010 = 01330010
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000acc
00059958 = 00000000
0005995c = 00000000

budimanjojo, 2020-01-18:
Manjaro Linux kernel 5.4.12-1
AMD Ryzen 5 3600
Gigabyte B450 AOURUS PRO

```
cat /sys/class/hwmon/hwmon0/debug_data
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01480020
0005a010 = 0113001f
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000b34
00059958 = 00000000
0005995c = 00000000
```

```
sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:           1.79 V  (min =  +0.00 V, max =  +2.78 V)
in1:           1.24 V  (min =  +0.00 V, max =  +2.78 V)
in2:           1.12 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:         3.36 V  (min =  +0.00 V, max =  +5.56 V)
in4:           1.26 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.11 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:          3.16 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +38.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.45 V
SVI2_SoC:      1.10 V
Tdie:         +54.4°C  (high = +70.0°C)
Tctl:         +54.4°C
Tccd1:        +53.2°C
SVI2_P_Core:  22.93 W
SVI2_P_SoC:   15.54 W
SVI2_C_Core:  15.81 A
SVI2_C_SoC:   14.13 A

it8686-isa-0a40
Adapter: ISA adapter
in0:           1.43 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.12 V  (min =  +0.00 V, max =  +3.06 V)
in5:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.12 V
fan1:        1298 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        2586 RPM  (min =   10 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:        1188 RPM  (min =   10 RPM)
temp1:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
temp2:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
temp3:        +55.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +41.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
temp6:        +45.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
intrusion0:  ALARM
```

SecretChief, 2020-01-19:
5.4.12-arch1-1 #1 SMP PREEMPT Tue, 14 Jan 2020 21:44:31 +0000 x86_64 GNU/Linux
ASUS Prime X399-A
TR 1950X

kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01490041
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000b5c
00059958 = 00000000
0005995c = 00000000

ocerman, 2020-01-19:
> 5.4.12-arch1-1 #1 SMP PREEMPT Tue, 14 Jan 2020 21:44:31 +0000 x86_64 GNU/Linux
> ASUS Prime X399-A
> TR 1950X
> 
> kernel_smn_support = 1
> 0005a008 = 00000002
> 0005a00c = 01490041
> 0005a010 = 01f70000
> 000598bc = 0fff00ff
> 0005994c = 00000000
> 00059954 = 00000b5c
> 00059958 = 00000000
> 0005995c = 00000000

@SecretChief  
Thanks for our output.

The 1950X should have 2 nodes, so zenpower should report for each node. Can you try to post also debug output for the other node? (it is also in `/sys/class/hwmon/hwmon[num]/debug_data`, but under different `num`, you can also try [bash script](https://gist.github.com/ocerman/6566b00072c8af96a109f65075f2d8b7) that should print values for both nodes)

Can you also post sensors output?

Lknechtli, 2020-01-20:
5.4.12-1-MANJARO
Ryzen 9 3900x
Gigabyte x570 Aorus Ultra

Debug data:
```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 014c0024
0005a010 = 01600006
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000a98
00059958 = 00000a78
0005995c = 00000000
00059960 = 00000000
```

Sensors:
```
it8792-isa-0a60
Adapter: ISA adapter
in0:           1.80 V  (min =  +0.00 V, max =  +2.78 V)
in1:         676.00 mV (min =  +0.00 V, max =  +2.78 V)
in2:           1.00 V  (min =  +0.00 V, max =  +2.78 V)
in3:           2.02 V  (min =  +0.00 V, max =  +2.78 V)
in4:           1.79 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.20 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          1.68 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:          1.54 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +37.0°C  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.48 V  
SVI2_SoC:      1.07 V  
Tdie:         +42.2°C  (high = +70.0°C)
Tctl:         +42.2°C  
Tccd1:        +33.5°C  
Tccd2:        +33.0°C  
SVI2_P_Core:  25.27 W  
SVI2_P_SoC:   11.07 W  
SVI2_C_Core:  16.47 A  
SVI2_C_SoC:   10.30 A
```

terroreek, 2020-01-20:
Gigabyte Aorus TRX40 Master
TR 3960x
Linux Kernel 5.4.12

```zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     7.00 mV 
SVI2_SoC:      1.08 V  
Tdie:         +54.5°C  (high = +70.0°C)
Tctl:         +54.5°C  
```
```kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 01f70000
0005a010 = 014c004f
000598bc = 0fff0fff
0005994c = 0fff0fff
00059954 = 00000ab6
00059958 = 00000000
0005995c = 00000ae8
```

jgslade, 2020-01-20:
Running Manjaro
```
System:    Kernel: 5.5.0-1-Navi14-Patch x86_64 bits: 64 compiler: gcc v: 9.2.0 Desktop: KDE Plasma 5.17.5 tk: Qt 5.14.0 
           wm: kwin_x11 dm: SDDM Distro: Manjaro Linux 
Machine:   Type: Desktop Mobo: BIOSTAR model: X370GT5 serial: <root required> UEFI: American Megatrends v: 5.13 
           date: 08/07/2018 
CPU:       Topology: 8-Core model: AMD Ryzen 7 1700 bits: 64 type: MT MCP arch: Zen rev: 1 L2 cache: 4096 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 114561 
           Speed: 3517 MHz min/max: 1550/3550 MHz boost: disabled Core speeds (MHz): 1: 3517 2: 3564 3: 3436 4: 3407 5: 3332 
           6: 3331 7: 3421 8: 3466 9: 3578 10: 3578 11: 3054 12: 3579 13: 2240 14: 1432 15: 2278 16: 1432
```
```
kernel_smn_support = 1
0005a008 = 00000002
0005a00c = 013c0018
0005a010 = 0149002b
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
```
```
SVI2_Core:     1.16 V  
SVI2_SoC:      1.09 V  
Tdie:         +46.5°C  (high = +70.0°C)
Tctl:         +46.5°C  
SVI2_P_Core:  93.22 W  
SVI2_P_SoC:   17.66 W  
SVI2_C_Core:  81.06 A  
SVI2_C_SoC:   16.23 A 
```

sitic, 2020-01-25:
Asus ROG Strix X399-E
TR 2950X
Arch, Linux Kernel 5.4.13
```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 01740027
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000b6e
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
kernel_smn_support = 1
node_id = 1
0005a008 = 00000002
0005a00c = 016d0000
0005a010 = 01f70000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
```

```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:          1.31 V  
CPU SOC Voltage:         828.00 mV 
DRAM AB Voltage:           1.16 V  
DRAM CD Voltage:           1.16 V  
1.8V PLL Voltage:          1.79 V  
+12V Voltage:             11.90 V  
+5V Voltage:               4.96 V  
3VSB Voltage:              3.27 V  
VBAT Voltage:              3.16 V  
AVCC3 Voltage:             3.29 V  
SB 1.05V Voltage:          1.06 V  
CPU Core Voltage:          1.29 V  
CPU SOC Voltage:         824.00 mV 
DRAM AB Voltage:           1.20 V  
DRAM CD Voltage:           1.20 V  
CPU Fan:                  750 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:              0 RPM
Chassis Fan 3:              0 RPM
Water Pump 1:               0 RPM
CPU OPT:                    0 RPM
EXT Fan 1:                  0 RPM
EXT Fan 2:                  0 RPM
EXT Fan 3:                  0 RPM
Cover Fan:                128 RPM
CPU Temperature:          +50.0°C  
CPU Socket Temperature:   +36.0°C  
Motherboard Temperature:  +34.0°C  
Chipset Temperature:      +53.0°C  
Tsensor Temperature:     +216.0°C  
CPU VRM Temperature:      +41.0°C  
EXT Tsensor 1:             +0.0°C  
EXT Tsensor 2:             +0.0°C  
EXT Tsensor 3:             +0.0°C  
CPU VRM Output Current:   56.00 A  

zenpower-pci-00cb
Adapter: PCI adapter
SVI2_Core:     1.32 V  
SVI2_SoC:      7.00 mV 
Tdie:         +46.4°C  (high = +95.0°C)
Tctl:         +73.4°C  
SVI2_P_Core:  72.65 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core:  56.12 A  
SVI2_C_SoC:    0.00 A  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:   825.00 mV 
SVI2_SoC:      7.00 mV 
Tdie:         +50.0°C  (high = +95.0°C)
Tctl:         +77.0°C  
SVI2_P_Core:  36.87 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core:  44.69 A  
SVI2_C_SoC:    0.00 A  
```

scorpio810, 2020-01-26:
Threadripper 3960X with a MSI TRX40 PRO 10G
Linux debian 5.4.14-vanilla #1 SMP Thu Jan 23 15:12:27 CET 2020 x86_64 GNU/Linux

```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 01f80000
0005a010 = 014b003a
000598bc = 0fff0fff
0005994c = 0fff0fff
00059954 = 00000a8c
00059958 = 00000000
0005995c = 00000a94
00059960 = 00000000
```

AndersDala, 2020-01-26:
5.3.0-26-generic
AMD Ryzen 7 1700
X370 GAMING PRO CARBON (MS-7A32)

kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 011d0029
0005a010 = 014b0024
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.37 V  
SVI2_SoC:     +1.08 V  
Tdie:         +41.0°C  (high = +95.0°C)
Tctl:         +41.0°C  
SVI2_P_Core:  40.01 W  
SVI2_P_SoC:   12.95 W  
SVI2_C_Core: +30.14 A  
SVI2_C_SoC:  +11.90 A  

SVI2_C_Core: +34.29 A  
SVI2_C_SoC:  +11.18 A

crt0mega, 2020-01-27:
Debian GNU/Linux Unstable; Kernel 5.4.8; ASUS ROG Strix GL702ZC (B350-based with Ryzen 1700)

### Debug data:

```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 01720008
0005a010 = 01640038
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
```

### Sensors:
```
crt0mega@curie:~/Projekte$ sensors zenpower-pci-00c3
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:   913.00 mV 
SVI2_SoC:    925.00 mV 
Tdie:         +50.8°C  (high = +95.0°C)
Tctl:         +50.8°C  
SVI2_P_Core:  15.38 W  
SVI2_P_SoC:   18.35 W  
SVI2_C_Core:  17.67 A  
SVI2_C_SoC:   19.84 A
```

scorpio810, 2020-01-28:
Initial threadripper/epyc support commit ;-)

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_SoC:     +1.07 V  
Tdie:         +66.8°C  (high = +95.0°C)
Tctl:         +66.8°C  
Tccd1:        +65.2°C  
SVI2_P_SoC:   28.47 W  
SVI2_C_SoC:  +25.31 A  

amdgpu-pci-2100
Adapter: PCI adapter
vddgfx:       +0.93 V  
fan1:         952 RPM  (min =    0 RPM, max = 3500 RPM)
edge:         +43.0°C  (crit = +97.0°C, hyst = -273.1°C)
power1:       11.15 W  (cap =  36.00 W)

nct6797-isa-0a20
Adapter: ISA adapter
in0:                    +1.26 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.56 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                  1412 RPM  (min =    0 RPM)
fan2:                  1187 RPM  (min =    0 RPM)
fan3:                   946 RPM  (min =    0 RPM)
fan4:                   828 RPM  (min =    0 RPM)
fan5:                   820 RPM  (min =    0 RPM)
fan6:                     0 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +31.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
CPUTIN:                 +40.0°C  (high = +112.0°C, hyst = +85.0°C)  sensor = thermistor
AUXTIN0:                +40.5°C  (high = +112.0°C, hyst = +85.0°C)  sensor = thermistor
AUXTIN1:                +44.0°C    sensor = thermistor
AUXTIN2:                +53.0°C    sensor = thermistor
AUXTIN3:                 -2.0°C    sensor = thermistor
SMBUSMASTER 0:          +66.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

vihu, 2020-01-29:
pc details:
```
CPU: Ryzen 3950x
Motherboard: Asus ROG Strix X570-I
Kernel: 5.3.18-1-MANJARO
```

debug_data:
```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 014b0030
0005a010 = 015c0004
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000aba
00059958 = 00000ab8
0005995c = 00000000
00059960 = 00000000
```

sensors zenpower-*:
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:   975.00 mV 
SVI2_SoC:      1.08 V  
Tdie:         +52.8°C  (high = +95.0°C)
Tctl:         +52.8°C  
Tccd1:        +39.2°C  
Tccd2:        +38.2°C  
SVI2_P_Core:   3.21 W  
SVI2_P_SoC:   15.28 W  
SVI2_C_Core:   3.95 A  
SVI2_C_SoC:   14.13 A
```

irusanov, 2020-01-31:
```bash
sudo ./zp_read_debug.sh 
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 0159002b
0005a010 = 014d000b
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000b0e
00059958 = 00000af8
0005995c = 00000000
00059960 = 00000000
```

```bash
sudo sensors zenpower-*
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.07 V  
SVI2_SoC:     +0.99 V  
Tdie:         +48.2°C  (high = +95.0°C)
Tctl:         +48.2°C  
Tccd1:        +49.8°C  
Tccd2:        +45.5°C  
SVI2_P_Core:   9.15 W  
SVI2_P_SoC:   12.29 W  
SVI2_C_Core:  +9.22 A  
SVI2_C_SoC:  +12.36 A 
```

```bash
uname -r
5.3.0-28-generic
```

SMU Debug Tool
```json
{
  "CpuId": "870F10",
  "CpuName": "AMD Ryzen 9 3900X 12-Core Processor",
  "MbVendor": "ASUSTeK COMPUTER INC.",
  "MbName": "CROSSHAIR VI HERO",
  "BiosVersion": "7704",
  "SmuVersion": "46.54.00",
  "Mailboxes": [
    {
      "MsgAddress": "0x03B10524",
      "RspAddress": "0x03B10570",
      "ArgAddress": "0x03B10A40"
    },
    {
      "MsgAddress": "0x03B10528",
      "RspAddress": "0x03B10574",
      "ArgAddress": "0x03B10A60"
    },
    {
      "MsgAddress": "0x03B10530",
      "RspAddress": "0x03B1057C",
      "ArgAddress": "0x03B109C4"
    }
  ]
}
```

spiicytuna, 2020-02-16:
Ryzen 3590
Crosshair VIII Impact
Kernel: 5.3.0-29-generic

```
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 014a0027
0005a010 = 015f0002
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000aca
00059958 = 00000ac2
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000
```

```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.96 V  
SVI2_SoC:     +1.09 V  
Tdie:         +50.5°C  (high = +95.0°C)
Tctl:         +50.5°C  
Tccd1:        +40.5°C  
Tccd2:        +39.5°C  
SVI2_P_Core:   2.52 W  
SVI2_P_SoC:   12.49 W  
SVI2_C_Core:  +2.63 A  
SVI2_C_SoC:  +11.48 A  
```

lizardb0y, 2020-02-16:
Kernel: 5.0.0-38-generic
Motherboard: Seco C40 Seco (Udoo Bolt V8)
CPU:  AMD Ryzen Embedded V1605B with Radeon Vega Gfx

kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 01600029
0005a010 = 0164000f
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00006e37
0005996c = 0000003f
00059970 = c0800005

load average: 7.45, 8.40, 7.37
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.95 V  
SVI2_SoC:     +0.92 V  
Tdie:         +54.2°C  (high = +95.0°C)
Tctl:         +54.2°C  
SVI2_P_Core:  59.23 W  
SVI2_P_SoC:    8.62 W  
SVI2_C_Core: +63.39 A  
SVI2_C_SoC:   +9.38 A

lizardb0y, 2020-02-16:
Using [this](https://gist.github.com/lizardb0y/edf7d7e7f8aae92d41fa1fb01d6c3dad#file-zp_read_debug-sh) extended bash script:

Kernel: 5.0.0-38-generic
Motherboard: Seco C40 Seco (Udoo Bolt V8)
CPU:  AMD Ryzen Embedded V1605B with Radeon Vega Gfx

kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 0173000b
0005a010 = 01640014
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00005e2f
0005996c = 0000003f
00059970 = c0800005

load average: 7.45, 8.40, 7.37
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.01 V  
SVI2_SoC:     +0.93 V  
Tdie:         +47.2°C  (high = +95.0°C)
Tctl:         +47.2°C  
SVI2_P_Core:  25.27 W  
SVI2_P_SoC:    5.34 W  
SVI2_C_Core: +24.94 A  
SVI2_C_SoC:   +5.77 A

klikzip, 2020-02-18:
Kernel: 5.4.19-200.fc31.x86_64
Motherboard: ASUSTeK COMPUTER INC. PRIME X470-PRO ASUSTeK COMPUTER INC.
CPU:  AMD Ryzen 5 3600 6-Core Processor

load average: 12,56, 10,49, 5,76
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.32 V  
SVI2_SoC:     +1.09 V  
Tdie:         +76.2°C  (high = +95.0°C)
Tctl:         +76.2°C  
Tccd1:        +76.2°C  
SVI2_P_Core:  42.77 W  
SVI2_P_SoC:   10.89 W  
SVI2_C_Core: +32.28 A  
SVI2_C_SoC:  +10.01 A  

kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 014a0020
0005a010 = 01240031
0005a014 = 00000000
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000bee
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000

harrykipper, 2020-02-19:
Processor: EPYC 7401
Motherboard: Supermicro H11DSi-NT
Kernel: 5.5.4

$ sensors zenpower-*
zenpower-pci-00eb
Adapter: PCI adapter
Tdie:         +30.2°C  (high = +95.0°C)
Tctl:         +30.2°C  

zenpower-pci-00db
Adapter: PCI adapter
Tdie:         +32.0°C  (high = +95.0°C)
Tctl:         +32.0°C  

zenpower-pci-00fb
Adapter: PCI adapter
Tdie:         +29.0°C  (high = +95.0°C)
Tctl:         +29.0°C  

zenpower-pci-00cb
Adapter: PCI adapter
SVI2_Core:   694.00 mV 
Tdie:         +31.6°C  (high = +95.0°C)
Tctl:         +31.6°C  
SVI2_P_Core:   0.00 W  
SVI2_C_Core:   0.00 A  

zenpower-pci-00f3
Adapter: PCI adapter
Tdie:         +29.5°C  (high = +95.0°C)
Tctl:         +29.5°C  

zenpower-pci-00e3
Adapter: PCI adapter
Tdie:         +30.2°C  (high = +95.0°C)
Tctl:         +30.2°C  
Tccd1:        +30.2°C  
Tccd2:        +29.8°C  
Tccd3:        +29.0°C  

zenpower-pci-00d3
Adapter: PCI adapter
Tdie:         +32.0°C  (high = +95.0°C)
Tctl:         +32.0°C  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_SoC:    944.00 mV 
Tdie:         +32.2°C  (high = +95.0°C)
Tctl:         +32.2°C  
Tccd1:        +32.2°C  
Tccd2:        +32.2°C  
Tccd3:        +32.0°C  
SVI2_P_SoC:   21.80 W  
SVI2_C_SoC:   23.09 A  

zp_read_debug.sh
kernel_smn_support = 1
node_id = 7
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00003a1d
0005996c = 0000003e
00059970 = c0800005
kernel_smn_support = 1
node_id = 0
0005a008 = 00000002
0005a00c = 01620040
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000a8a
00059958 = 00000a8e
0005995c = 00000a90
00059960 = 00000000
00059964 = 08400001
00059968 = 00003e1f
0005996c = 00000040
00059970 = c0800005
kernel_smn_support = 1
node_id = 1
0005a008 = 00000002
0005a00c = 018a0000
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00003e1f
0005996c = 00000041
00059970 = c0800005
kernel_smn_support = 1
node_id = 2
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00004020
0005996c = 00000043
00059970 = c0800005
kernel_smn_support = 1
node_id = 3
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00003e1f
0005996c = 00000042
00059970 = c0800005
kernel_smn_support = 1
node_id = 4
0005a008 = 00000002
0005a00c = 01620043
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000a7a
00059958 = 00000a76
0005995c = 00000a74
00059960 = 00000000
00059964 = 08400001
00059968 = 00003a1d
0005996c = 0000003f
00059970 = c0800005
kernel_smn_support = 1
node_id = 5
0005a008 = 00000002
0005a00c = 018a0000
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00003c1e
0005996c = 00000043
00059970 = c0800005
kernel_smn_support = 1
node_id = 6
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00003a1d
0005996c = 00000042
00059970 = c0800005

joy4eg, 2020-02-21:
3900x, x570 aorus master
```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 014e0027
0005a010 = 0160000b
0005a014 = 00000000
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000ad4
00059958 = 00000ad0
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000
```
```
it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.68 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +0.99 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +1.67 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.79 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.19 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.67 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.62 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         734 RPM  (min =    0 RPM)
temp1:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.45 V  
SVI2_SoC:     +1.06 V  
Tdie:         +54.1°C  (high = +95.0°C)
Tctl:         +54.1°C  
Tccd1:        +40.0°C  
Tccd2:        +40.5°C  
SVI2_P_Core:  29.76 W  
SVI2_P_SoC:   12.44 W  
SVI2_C_Core: +19.76 A  
SVI2_C_SoC:  +11.77 A  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +53.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +80.8°C)
Sensor 1:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +63.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +50.9°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +53.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +80.8°C)
Sensor 1:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +65.8°C  (low  = -273.1°C, high = +65261.8°C)

```

dobb13, 2020-02-24:
- Zenpower from arch user repo 44.ee05fcf-1
- Manjaro Linux kernel 5.5.2-1
- Ryzen 2600X
- ASUS TUF B350M-PLUS GAMING

SoC and CPU info are correct
```
~ >>> cat /sys/class/hwmon/hwmon2/debug_data                                   
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 0161000d
0005a010 = 0141001e
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00004221
0005996c = 0000004b
00059970 = c0800005
~ >>> sensors zenpower-pci-00c3                                                
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:   944.00 mV 
SVI2_SoC:      1.14 V  
Tdie:         +33.9°C  (high = +95.0°C)
Tctl:         +33.9°C  
SVI2_P_Core:   9.81 W  
SVI2_P_SoC:   11.08 W  
SVI2_C_Core:  10.39 A  
SVI2_C_SoC:    9.74 A  
```

leogx9r, 2020-02-24:
Processor: Ryzen 7 3700X w/ PBO
Motherboard: MSI B450 Tomahawk MAX
OS: Arch Linux
Kernel: 5.5.5
Version: Latest commit ( ee05fcf )

Debug Data: (
```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 014b001c
0005a010 = 01630003
0005a014 = 00000000
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000aa0
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000
```

Sensor Output (Under stress test):
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.34 V  
SVI2_SoC:      1.09 V  
Tdie:         +72.4°C  (high = +95.0°C)
Tctl:         +72.4°C  
Tccd1:        +72.2°C  
SVI2_P_Core:  65.23 W  
SVI2_P_SoC:    9.93 W  
SVI2_C_Core:  48.75 A  
SVI2_C_SoC:    9.12 A
```

Sensor Output of Motherboard (nct6796):

```
nct6797-isa-0a20
Adapter: ISA adapter
Vcore:                   1.34 V  (min =  +0.10 V, max =  +1.50 V)
+5V:                     4.96 V  (min =  +4.76 V, max =  +5.24 V)
AVCC:                    3.36 V  (min =  +3.17 V, max =  +3.44 V)
3VCC:                    3.36 V  (min =  +3.17 V, max =  +3.44 V)
+12V:                   12.00 V  (min = +11.42 V, max = +12.58 V)
3VSB:                    3.36 V  (min =  +3.17 V, max =  +3.44 V)
VBAT:                    3.31 V  (min =  +3.17 V, max =  +3.44 V)
CPU +1.8V:               1.82 V  (min =  +1.77 V, max =  +1.84 V)
SoC:                     1.10 V  (min =  +1.00 V, max =  +1.20 V)
DIMM:                    1.38 V  (min =  +1.20 V, max =  +1.46 V)
```

mdlayher, 2020-02-25:
Processor: Ryzen 9 3900X
Motherboard: ASUS ROG Strix X570-E Gaming
OS: Ubuntu 19.10
Kernel: 5.3.0-40-generic (Ubuntu)
Version: 0.1.10

Debug:
```
$ ./zp_read_debug.sh 
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 014b0031
0005a010 = 010f0013
0005a014 = 00000000
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000aca
00059958 = 00000ad4
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000
```

Sensors:

```
$ sensors zenpower-*
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.96 V  
SVI2_SoC:     +1.08 V  
Tdie:         +47.8°C  (high = +95.0°C)
Tctl:         +47.8°C  
Tccd1:        +37.8°C  
Tccd2:        +39.0°C  
SVI2_P_Core:   3.78 W  
SVI2_P_SoC:   15.28 W  
SVI2_C_Core:  +3.95 A  
SVI2_C_SoC:  +14.13 A  
```

Thanks so much for this module! I'll update to the latest version on my other Ryzen box and report on that one shortly too.

mdlayher, 2020-02-25:
Processor: Ryzen 7 2700
Motherboard: Gigabyte AX370-Gaming K7
OS: Ubuntu 18.04 LTS
Kernel: 4.15.0-88-generic (Ubuntu)
Version: 0.1.10

Debug:

```
$ ./zp_read_debug.sh 
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 017a0003
0005a010 = 01750011
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 0000381c
0005996c = 00000035
00059970 = c0800005
```

Sensors:

```
$ sensors zenpower-*
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.79 V  
SVI2_SoC:     +0.82 V  
Tdie:         +28.0°C  (high = +95.0°C)
Tctl:         +28.0°C  
SVI2_P_Core:   2.46 W  
SVI2_P_SoC:    5.02 W  
SVI2_C_Core:  +5.20 A  
SVI2_C_SoC:   +6.13 A 
```

abucodonosor, 2020-02-27:
AMD Ryzen 7 3750H
Linux 5.5.6
Acer Nitro5 ( AN515-43 )

```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 01700005
0005a010 = 01780013
0005a014 = 00000000
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00005c2e
0005996c = 0000005d
00059970 = c0800005

zenpower-pci-00c3   
Adapter: PCI adapter  
SVI2_Core:     1.17 V
SVI2_SoC:    982.00 mV
Tdie:         +87.1°C  (high = +95.0°C)
Tctl:         +87.1°C
SVI2_P_Core:  55.88 W
SVI2_P_SoC:   10.27 W
SVI2_C_Core:  47.80 A
SVI2_C_SoC:   10.46 A 

```

abucodonosor, 2020-02-27:
AMD Ryzen 3 2200G
Linux 5.5.6
ASRock DeskMini A300

```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 011d001b
0005a010 = 0156000f
0005a014 = 00000000
000598bc = ffffffff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00005028
0005996c = 0000004c
00059970 = c0800005

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.32 V
SVI2_SoC:      1.09 V
Tdie:         +71.4°C  (high = +95.0°C)
Tctl:         +71.4°C
SVI2_P_Core:  95.01 W <-- that looks wrong
SVI2_P_SoC:    7.89 W
SVI2_C_Core:  71.70 A
SVI2_C_SoC:    7.21 A
```

abucodonosor, 2020-02-27:
AMD EPYC 7281 * 2
Linux 5.5.6
Supermicro H11DSi-NT

```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 4
0005a008 = 00000002
0005a00c = 01610038
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000a3c
00059958 = 00000a3a
0005995c = 00000a3e
00059960 = 00000000
00059964 = 08400001
00059968 = 00003018
0005996c = 0000002c
00059970 = c0800005
KERN_SUP: 1
NODE1; CPU0; N/CPU: 4
0005a008 = 00000002
0005a00c = 01880000
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002c16
0005996c = 0000002b
00059970 = c0800005
KERN_SUP: 1
NODE2; CPU0; N/CPU: 4
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002a15
0005996c = 0000002b
00059970 = c0800005
KERN_SUP: 1
NODE3; CPU0; N/CPU: 4
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002c16
0005996c = 0000002a
00059970 = c0800005
KERN_SUP: 1
NODE4; CPU1; N/CPU: 4
0005a008 = 00000002
0005a00c = 01620040
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000a2e
00059958 = 00000a20
0005995c = 00000a28
00059960 = 00000000
00059964 = 08400001
00059968 = 00002613
0005996c = 00000026
00059970 = c0800005
KERN_SUP: 1
NODE5; CPU1; N/CPU: 4
0005a008 = 00000002
0005a00c = 01660001
0005a010 = 01f70000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002814
0005996c = 00000028
00059970 = c0800005
KERN_SUP: 1
NODE6; CPU1; N/CPU: 4
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002412
0005996c = 00000024
00059970 = c0800005
KERN_SUP: 1
NODE7; CPU1; N/CPU: 4
0005a008 = 00000002
0005a00c = 00000000
0005a010 = 00000000
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00002613
0005996c = 00000025
00059970 = c0800005

enpower-pci-00f3
Adapter: PCI adapter
cpu1 Tdie:    +26.0°C  (high = +95.0°C)
cpu1 Tctl:    +26.0°C  

zenpower-pci-00e3
Adapter: PCI adapter
cpu1 SVI2_SoC:   938.00 mV 
cpu1 Tdie:        +30.5°C  (high = +95.0°C)
cpu1 Tctl:        +30.5°C  
cpu1 Tccd1:       +31.2°C  
cpu1 Tccd2:       +26.5°C  
cpu1 Tccd3:       +26.8°C  
cpu1 SVI2_P_SoC:  27.07 W  
cpu1 SVI2_C_SoC:  28.86 A  

zenpower-pci-00d3
Adapter: PCI adapter
cpu0 Tdie:    +32.0°C  (high = +95.0°C)
cpu0 Tctl:    +32.0°C  

zenpower-pci-00c3
Adapter: PCI adapter
cpu0 SVI2_SoC:   938.00 mV 
cpu0 Tdie:        +35.0°C  (high = +95.0°C)
cpu0 Tctl:        +35.0°C  
cpu0 Tccd1:       +31.8°C  
cpu0 Tccd2:       +32.0°C  
cpu0 Tccd3:       +31.5°C  
cpu0 SVI2_P_SoC:  26.40 W  
cpu0 SVI2_C_SoC:  28.50 A  

zenpower-pci-00fb
Adapter: PCI adapter
cpu1 Tdie:    +26.4°C  (high = +95.0°C)
cpu1 Tctl:    +26.4°C  

zenpower-pci-00eb
Adapter: PCI adapter
cpu1 SVI2_Core:   969.00 mV 
cpu1 Tdie:         +30.5°C  (high = +95.0°C)
cpu1 Tctl:         +30.5°C  
cpu1 SVI2_P_Core:  43.30 W  
cpu1 SVI2_C_Core:  45.73 A  

zenpower-pci-00db
Adapter: PCI adapter
cpu0 Tdie:    +31.6°C  (high = +95.0°C)
cpu0 Tctl:    +31.6°C  

zenpower-pci-00cb
Adapter: PCI adapter
cpu0 SVI2_Core:   957.00 mV 
cpu0 Tdie:         +31.8°C  (high = +95.0°C)
cpu0 Tctl:         +31.8°C  
cpu0 SVI2_P_Core:  46.74 W  
cpu0 SVI2_C_Core:  47.80 A  

```

vvrein, 2020-03-04:
### CPU: AMD Ryzen 5 3600, MB: GA-A320MA-M.2
```

# uname -a
Linux tokk 5.5.7-arch1-1 #1 SMP PREEMPT Sat, 29 Feb 2020 19:06:02 +0000 x86_64 GNU/Linux

# cat /sys/class/hwmon/hwmon1/debug_data
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 0155001f
0005a010 = 01640007
0005a014 = 00000000
000598bc = 0fff0fff
0005994c = 00000000
00059954 = 00000ae4
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 00000000
00059968 = 00000000
0005996c = 00000000
00059970 = 00000000

# sensors zenpower-*
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:   925.00 mV
SVI2_SoC:      1.02 V
Tdie:         +44.0°C  (high = +95.0°C)
Tctl:         +44.0°C
Tccd1:        +45.0°C
SVI2_P_Core:   4.87 W
SVI2_P_SoC:    9.24 W
SVI2_C_Core:   5.27 A
SVI2_C_SoC:    9.12 A
```

jfrob, 2020-03-05:
Ryzen 2700x
ASUS Prime x470-Pro
Kernel 5.5.6-ck-r1

```
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 01230022
0005a010 = 01580020
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00006432
0005996c = 0000003f
00059970 = c0800005
```

Normal unidle load:
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.34 V  
SVI2_SoC:     +1.00 V  
Tdie:         +42.1°C  (high = +95.0°C)
Tctl:         +52.1°C  
SVI2_P_Core:  43.30 W  
SVI2_P_SoC:   11.18 W  
SVI2_C_Core: +35.33 A  
SVI2_C_SoC:  +11.18 A
```

7z b:
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.31 V  
SVI2_SoC:     +1.00 V  
Tdie:         +53.5°C  (high = +95.0°C)
Tctl:         +63.5°C  
SVI2_P_Core: 100.51 W  
SVI2_P_SoC:   14.43 W  
SVI2_C_Core: +76.90 A  
SVI2_C_SoC:  +14.43 A
```

maximlevitsky, 2020-03-14:
KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 01f70000
0005a010 = 01530039
0005a014 = 01110012
000598bc = 0fff0fff
0005994c = 0fff0fff
00059954 = 00000ac4
00059958 = 00000000
0005995c = 00000aaa
00059960 = 00000000
00059964 = 00000aae
00059968 = 00000000
0005996c = 00000ab4
00059970 = 00000000

Threadripper 3970X / Gigabyte Desinare TRX40

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.97 V  
SVI2_SoC:     +1.03 V  
Tdie:         +49.8°C  (high = +95.0°C)
Tctl:         +49.8°C  
Tccd1:        +39.5°C  
Tccd3:        +35.8°C  
Tccd5:        +38.2°C  
Tccd7:        +38.2°C  
SVI2_P_Core:   7.66 W  
SVI2_P_SoC:   17.31 W  
SVI2_C_Core:  +7.91 A  
SVI2_C_SoC:  +16.77 A  

I strongly suspect that core power is 1/2 of the real value.
When I run prime95 with the highest power settings (torture test 2), this is what I see

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.99 V  
SVI2_SoC:     +1.03 V  
Tdie:         +72.0°C  (high = +95.0°C)
Tctl:         +72.0°C  
Tccd1:        +72.2°C  
Tccd3:        +63.5°C  
Tccd5:        +71.0°C  
Tccd7:        +65.5°C  
SVI2_P_Core: 123.02 W  
SVI2_P_SoC:   18.53 W  
SVI2_C_Core: +123.86 A  
SVI2_C_SoC:  +17.95 A  

In windows the core power usage is above 200W.

Chlorophytus, 2020-04-03:
> KERN_SUP: 1
> NODE0; CPU0; N/CPU: 1
> 0005a008 = 00000002
> 0005a00c = 01f70000
> 0005a010 = 01530039
> 0005a014 = 01110012
> 000598bc = 0fff0fff
> 0005994c = 0fff0fff
> 00059954 = 00000ac4
> 00059958 = 00000000 [...]

Should probably just be a simple bitshift. I'd have to check how precise the sensors read out (in an AMD MSR Developer Guide, etc.) before I put in a `<< 1` or similar in a fork. Maybe.

[98 more replies]