ocerman/zenpower #16: Different Output of SV12_Core from Zenmonitor

budimanjojo, 2020-01-19:
Hello, I'm using Ryzen 5 3600 with Gigabyte B450 AORUS PRO motherboard. I have a conky script displaying the output on my desktop and my SV12_Core stays at 1.4 V constantly (not going down even on 2.5GHz idle). And I installed zenmonitor and I've got different result. In zenmonitor, the CPU Core Voltage (SV12) goes up and down depending on the CPU frequency (1.1 V on idle and 1.4 V on boost). So why is the core voltage displaying differently? Here's a screenshot showing it.
[image]

ocerman, 2020-01-19:
Hello,

What kind of cpu governor are you using? It is possible that your CPU governor is too aggressive and the conky/sensors update function is causing enough load for CPU to rise voltage and clocks. 

I had similar problem with hardinfo (issue https://github.com/lpereira/hardinfo/issues/312), but managed to change the code to be more effective and do not cause this issue anymore (https://github.com/lpereira/hardinfo/pull/327)

Anyway - the zenmonitor's update function is as lightweight as possible, that's why this is not an issue there.

Can you tell me how are you reading sensor values to conky?
I think that best way would be to read raw values from `/sys/class/hwmon/hwmon[num]/`

If you are parsing sensors output, you can try watch only zenpower: `watch -n 0.5 sensors zenpower-*`

Let me know if it helps.

budimanjojo, 2020-01-19:
I was using ondemand cpu governor. And this is what I have for the conky:
```
${color2}CPU Core Voltage ${alignr}${color0}${hwmon 0 vol 1} V
```
And actually if I changed the governor to `conservative` then it's showing right. What's weird is zenmonitor always showing the right value but the conky doesn't show the right value with ondemand. Maybe you are right about conky hwmon command being not effective or there's something wrong with ondemand governor.

ocerman, 2020-01-19:
Try this:
```
${color2}CPU Core Voltage ${alignr}${color0}${exec awk '{printf "%3.2fV", $1/1000}' /sys/class/hwmon/hwmon0/in1_input }
```

budimanjojo, 2020-01-19:
It actually works great now, thanks!

ocerman, 2020-01-19:
np

closed 2020-01-19 by ocerman