ocerman/zenpower #18: Better support for Threadripper 3000

ocerman, 2020-01-20:
Based on @terroreek's report https://github.com/ocerman/zenpower/issues/12#issuecomment-576105427

Implement:
- [ ] Core voltage
- [ ] Core current/power
- [ ] SOC current/power
- [ ] CCD temperatures

ocerman, 2020-02-15:
@scorpio810 @Chlorophytus @terroreek
The last commit should bring SVI2 Core values. Can any Threadripper 3000 user confirm that?

Compizfox, 2020-02-15:
Seems to work okay!

     zenpower-pci-00c3
     Adapter: PCI adapter
     SVI2_Core:     1.40 V
     SVI2_SoC:      1.09 V
     Tdie:         +52.0°C  (high = +95.0°C)
     Tctl:         +52.0°C
     Tccd1:        +51.2°C
     Tccd3:        +51.8°C
     Tccd5:        +51.0°C
     Tccd7:        +49.0°C
     SVI2_P_Core: 102.38 W
     SVI2_P_SoC:   14.49 W
     SVI2_C_Core:  73.13 A
     SVI2_C_SoC:   13.54 A

Chlorophytus, 2020-02-16:
Seems to work, except for SVI2 P/C Core but it may be my motherboard settings. (TR 3960X)
```
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.72 V  
SVI2_SoC:     +1.04 V  
Tdie:         +31.1°C  (high = +95.0°C)
Tctl:         +31.1°C  
Tccd1:        +30.5°C  
Tccd3:        +31.2°C  
Tccd5:        +30.5°C  
Tccd7:        +27.2°C  
SVI2_P_Core:   0.00 W  
SVI2_P_SoC:   18.33 W  
SVI2_C_Core:  +0.00 A  
SVI2_C_SoC:  +17.66 A  
```

scorpio810, 2020-02-16:
zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.95 V  
SVI2_SoC:     +1.08 V  
Tdie:         +41.8°C  (high = +95.0°C)
Tctl:         +41.8°C  
Tccd1:        +29.2°C  
Tccd3:        +31.0°C  
Tccd5:        +31.5°C  
Tccd7:        +30.8°C  
SVI2_P_Core:   3.13 W  
SVI2_P_SoC:   18.15 W  
SVI2_C_Core:  +3.29 A  
SVI2_C_SoC:  +17.07 A

ocerman, 2020-02-16:
Thanks everyone for testing.

Seems to be working fine except that current/power reading may not be accurate as that depends on the motherboard.

I will close this issue and tag new release soon.

closed 2020-02-16 by ocerman