ocerman/zenpower #19: Compilation errors on 0.1.6

terencode, 2020-01-24:
This is on Arch Linux with GCC 9.2:

```                                                                   
/var/lib/dkms/zenpower/30.c4e7e6c/build/make.log                                                                                          
DKMS make.log for zenpower-30.c4e7e6c for kernel 5.4.14-arch1-1 (x86_64)
ven. janv. 24 12:59:32 CET 2020
make: Entering directory '/usr/lib/modules/5.4.14-arch1-1/build'
  CC [M]  /var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.o
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c: In function ‘zenpower_read’:
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c:232:20: error: expected ‘)’ before ‘__attribute__’
  232 |    __attribute__ ((fallthrough)); // see note at zenpower_info
      |                    ^
      |                    )
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c:232:32: error: expected identifier or ‘(’ before ‘)’ token
  232 |    __attribute__ ((fallthrough)); // see note at zenpower_info
      |                                ^
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c:232:4: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  232 |    __attribute__ ((fallthrough)); // see note at zenpower_info
      |    ^~~~~~~~~~~~~
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c:231:12: warning: this statement may fall through [-Wimplicit-fallthrough=]
  231 |    channel -= 1;     // hwmon_in have different indexing,
      |    ~~~~~~~~^~~~
/var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.c:234:3: note: here
  234 |   case hwmon_curr:
      |   ^~~~
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/zenpower/30.c4e7e6c/build/zenpower.o] Error 1
make: *** [Makefile:1652: /var/lib/dkms/zenpower/30.c4e7e6c/build] Error 2
make: Leaving directory '/usr/lib/modules/5.4.14-arch1-1/build'
```

ocerman, 2020-01-24:
@terencode
Thx for report,
can you try it again?

terencode, 2020-01-24:
Compiles again, thanks!

closed 2020-01-24 by terencode