ocerman/zenpower #22: Can't compile latest version

akw28888, 2020-02-03:
Hi,

I can not compile the latest version of zenpower on my system.
0.1.5 works, but 0.1.6 & 0.1.7 failed.
I'm using CentOS 7, I tried GCC 4.8 & 7.3.1

```
The error log:
  CC [M]  /root/zenpower/zenpower.o
/root/zenpower/zenpower.c:379:2: error: implicit declaration of function ‘HWMON_CHANNEL_INFO’; did you mean ‘HWMON_C_LABEL’? [-Werror=implicit-function-declaration]
  HWMON_CHANNEL_INFO(temp,
  ^~~~~~~~~~~~~~~~~~
  HWMON_C_LABEL
/root/zenpower/zenpower.c:379:21: error: ‘temp’ undeclared here (not in a function); did you mean ‘sem’?
  HWMON_CHANNEL_INFO(temp,
                     ^~~~
                     sem
/root/zenpower/zenpower.c:385:21: error: ‘in’ undeclared here (not in a function); did you mean ‘inl’?
  HWMON_CHANNEL_INFO(in,
                     ^~
                     inl
/root/zenpower/zenpower.c:393:21: error: ‘curr’ undeclared here (not in a function); did you mean ‘char’?
  HWMON_CHANNEL_INFO(curr,
                     ^~~~
                     char
/root/zenpower/zenpower.c:397:21: error: ‘power’ undeclared here (not in a function)
  HWMON_CHANNEL_INFO(power,
                     ^~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:310: /root/zenpower/zenpower.o] Error 1
make[1]: *** [Makefile:1526: _module_/root/zenpower] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.19.94-300.el7.x86_64'
make: *** [Makefile:27: modules] Error 2
```

scorpio810, 2020-02-03:
Compile fine here 
```
 laurent@debian:/media/backup6/download/zenpower$ git pull
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 8 (delta 1), reused 4 (delta 1), pack-reused 0
Dépaquetage des objets: 100% (8/8), 12.56 Kio | 536.00 Kio/s, fait.
Depuis https://github.com/ocerman/zenpower
   5f3ac41..7eadfa9  master     -> origin/master
 * [nouvelle étiquette] v0.1.7     -> v0.1.7
Mise à jour 5f3ac41..7eadfa9
Fast-forward
 LICENSE    | 339 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 Makefile   |   2 +-
 zenpower.c |  47 ++++++++++++++++++++++++++++++++++-------------
 3 files changed, 374 insertions(+), 14 deletions(-)
 create mode 100644 LICENSE
[19:58:56] laurent@debian:/media/backup6/download/zenpower$ sudo make dkms-install
mkdir /usr/src/zenpower-0.1.7
cp /media/backup6/download/zenpower/dkms.conf /usr/src/zenpower-0.1.7
cp /media/backup6/download/zenpower/Makefile /usr/src/zenpower-0.1.7
cp /media/backup6/download/zenpower/zenpower.c /usr/src/zenpower-0.1.7
sed -e "s/@CFLGS@//" \
    -e "s/@VERSION@/0.1.7/" \
    -i /usr/src/zenpower-0.1.7/dkms.conf
dkms add zenpower/0.1.7

Creating symlink /var/lib/dkms/zenpower/0.1.7/source ->
                 /usr/src/zenpower-0.1.7

DKMS: add completed.
dkms build zenpower/0.1.7

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j48 KERNELRELEASE=5.5.1-vanilla TARGET=5.5.1-vanilla CFLAGS_MODULE+=...
cleaning build area...

DKMS: build completed.
dkms install zenpower/0.1.7

zenpower.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.5.1-vanilla/updates/dkms/

depmod...

DKMS: install completed.
[19:59:09] laurent@debian:/media/backup6/download/zenpower$ 
```

ocerman, 2020-02-03:
The `HWMON_CHANNEL_INFO` macro is not available on older kernels. I have pushed new commit 5d68cd732825115f359c5e5c5de18a63030e4f93 that should fix it. @akw28888 can you try it?

akw28888, 2020-02-04:
@ocerman 
I can compile it now.
Thank you.

closed 2020-02-04 by ocerman