ocerman/zenpower #23: Errors compiling on kernel 5.4 on ubuntu 18.04

gsedej, 2020-02-10:
hello

I am trying to build zenpower on ubuntu 18.04 with kernel 5.4 (installed via ukuu)

make output
```
$ sudo make dkms-install
mkdir /usr/src/zenpower-0.1.8
cp /home/gasper/git/zenpower/dkms.conf /usr/src/zenpower-0.1.8
cp /home/gasper/git/zenpower/Makefile /usr/src/zenpower-0.1.8
cp /home/gasper/git/zenpower/zenpower.c /usr/src/zenpower-0.1.8
sed -e "s/@CFLGS@//" \
    -e "s/@VERSION@/0.1.8/" \
    -i /usr/src/zenpower-0.1.8/dkms.conf
dkms add zenpower/0.1.8

Creating symlink /var/lib/dkms/zenpower/0.1.8/source ->
                 /usr/src/zenpower-0.1.8

DKMS: add completed.
dkms build zenpower/0.1.8

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.4.18-050418-generic TARGET=5.4.18-050418-generic CFLAGS_MODULE+=...(bad exit status: 2)
ERROR (dkms apport): binary package for zenpower: 0.1.8 not found
Error! Bad return status for module build on kernel: 5.4.18-050418-generic (x86_64)
Consult /var/lib/dkms/zenpower/0.1.8/build/make.log for more information.
Makefile:33: recipe for target 'dkms-install' failed
make: *** [dkms-install] Error 10
```

the log says 
```
././include/linux/compiler_types.h:210:24: error: expected ‘(’ before ‘__inline’
./arch/x86/include/asm/alternative.h:128:2: error: expected identifier or ‘(’ before string constant
etc..
```

full log [make.log](https://github.com/ocerman/zenpower/files/4181483/make.log)

ocerman, 2020-02-11:
Hello, I have found the problem: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1849348

> The linux headers from http://kernel.ubuntu.com/~kernel-ppa/mainline are built with a gcc 9 version. This causes CONFIG_CC_HAS_ASM_INLINE to be set to true.
> 
> However the gcc level that is installed by default on ubuntu 18.04.3 is still gcc 7.4. This level doesn't meet the criteria for supporting CONFIG_CC_HAS_ASM_INLINE, so the building with the linux headers will fail.
> 

You can either try to set CONFIG_CC_HAS_ASM_INLINE to false: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1849348/comments/6

> manually set the CONFIG_CC_HAS_ASM_INLINE statements false in the linux 5.4 headers (deleted the line in include/generated/autoconf.h and set CONFIG_CC_HAS_ASM_INLINE=n in include/config/auto.conf).

or, install newer gcc (version 8 should work fine): https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1849348/comments/10

> sudo apt install gcc-8 g++-8
> sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 80 --slave /usr/bin/g++ g++ /usr/bin/g++-8 --slave /usr/bin/gcov gcov /usr/bin/gcov-8

gsedej, 2020-02-13:
THANK YOU very much for research and instructions.
I can confirm both methods work.

### 1) use old gcc 7.4 and edit CONFIG_CC_HAS_ASM_INLINE
my kernel is 5.6.0-050600rc1
delete 
```
#define CONFIG_CC_HAS_ASM_INLINE 0
```
in
```
/usr/src/linux-headers-5.6.0-050600rc1-generic/include/generated/autoconf.h
```
and set
```
CONFIG_CC_HAS_ASM_INLINE=y
```
to
```
CONFIG_CC_HAS_ASM_INLINE=n
```
in
```
/usr/src/linux-headers-5.6.0-050600rc1-generic/include/config/auto.conf
```

### 2) install gcc8 and set alternatives
```
sudo apt install gcc-8 g++-8
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 80 --slave /usr/bin/g++ g++ /usr/bin/g++-8 --slave /usr/bin/gcov gcov /usr/bin/gcov-8
sudo make dkms-uninstall
sudo make dkms-install
```

anyway this are merged output of k10temp (kernel 5.6) and zenpower
```
k10temp-pci-00c3
Adapter: PCI adapter
Vcore:        +0.95 V  
Vsoc:         +1.09 V  
Tdie:         +37.6°C  
Tctl:         +37.6°C  
Tccd1:        +37.5°C  
Icore:        +5.00 A  
Isoc:         +6.50 A  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +0.95 V  
SVI2_SoC:     +1.09 V  
Tdie:         +37.5°C  (high = +95.0°C)
Tctl:         +37.5°C  
Tccd1:        +37.8°C  
SVI2_P_Core:   2.50 W  
SVI2_P_SoC:    8.32 W  
SVI2_C_Core:  +2.63 A  
SVI2_C_SoC:   +7.65 A 
```

should this be mentioned somewhere? I am using "old" distros with new kernels (and new 3D userspace drivers). I don't know if there are many others using like this.

(updating kernel is super easy via ukuu and 3D drivers via PPA)

ocerman, 2020-02-15:
I have updated the readme file

closed 2020-02-15 by ocerman