ocerman/zenpower #25: Zen power can't report each core's temperature.

diodep, 2020-03-04:
My mainboard is ASUS PRO WS X570 ACE. I have compiled zenpower (on Linux kernel v5.5.7). But it only shows Tccd, Tdie, Tctl. Each core's temperature doesn't show up in sensors application.

> hikari@kanau:~$ sensors
> nct6798-isa-0290
> Adapter: ISA adapter
> in0:                    +1.24 V  (min =  +0.00 V, max =  +1.74 V)
> in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in2:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in3:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in4:                    +1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in5:                    +0.59 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in6:                    +1.09 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in7:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in8:                    +3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in9:                    +0.90 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in10:                   +0.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in11:                   +0.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in12:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in13:                   +1.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in14:                   +0.90 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> fan1:                  2368 RPM  (min =    0 RPM)
> fan2:                  1790 RPM  (min =    0 RPM)
> fan3:                  1744 RPM  (min =    0 RPM)
> fan4:                     0 RPM  (min =    0 RPM)
> fan7:                     0 RPM  (min =    0 RPM)
> SYSTIN:                 +18.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
> CPUTIN:                 +23.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
> AUXTIN0:                +21.5°C    sensor = thermistor
> AUXTIN1:               +127.0°C    sensor = thermistor
> AUXTIN2:               +123.0°C    sensor = thermistor
> AUXTIN3:                +31.0°C    sensor = thermistor
> PCH_CHIP_CPU_MAX_TEMP:   +0.0°C
> PCH_CHIP_TEMP:           +0.0°C
> PCH_CPU_TEMP:            +0.0°C
> PCH_MCH_TEMP:            +0.0°C
> intrusion0:            ALARM
> intrusion1:            OK
> beep_enable:           disabled
> 
> nouveau-pci-0800
> Adapter: PCI adapter
> fan1:           0 RPM
> temp1:        +28.0°C  (high = +95.0°C, hyst =  +3.0°C)
>                        (crit = +105.0°C, hyst =  +5.0°C)
>                        (emerg = +135.0°C, hyst =  +5.0°C)
> 
> zenpower-pci-00c3
> Adapter: PCI adapter
> SVI2_Core:    +1.25 V
> SVI2_SoC:     +1.09 V
> Tdie:         +23.5°C  (high = +95.0°C)
> Tctl:         +23.5°C
> Tccd1:        +22.8°C
> Tccd2:        +23.2°C
> SVI2_P_Core:  13.18 W
> SVI2_P_SoC:   10.25 W
> SVI2_C_Core: +11.20 A
> SVI2_C_SoC:   +9.42 A
> 
> hikari@kanau:~$
>

ocerman, 2020-03-07:
Hello, thanks for your report.

Unfortunatelly the Ryzen CPUs do not have sensors for each core.

closed 2020-03-07 by ocerman

diodep, 2020-03-09:
Thank you for your answer! I mistakenly ask the question.