ocerman/zenpower #26: R5 2600

G4E, 2020-03-09:
Hello, I have a R5 2600, but it seems that zenpower doesn't recognize it ?
Config :
CPU : Ryzen 5 2600
MotherBoard : Asrock B450m PRO4
Distribution : ArchLinux
Kernel : 5.5.8-zen1-1-zen (Zen Kernel)
```
[G4E@G4E-pc ~]$ lsmod | grep zenpower
zenpower               16384  0
```
```
[G4E@G4E-pc ~]$ sudo zp_read_debug.sh 
Zenpower not found
```
```
[G4E@G4E-pc ~]$ sensors zenpower-*
Specified sensor(s) not found!
```

ocerman, 2020-03-09:
Can you try reboot? If that does not help, can you post output of `lspci`?

G4E, 2020-03-09:
I just rebooted. Something interesting : 
```
[tom@tom-pc ~]$ lsmod | grep zenpower
zenpower               16384  0
[tom@tom-pc ~]$ lsmod | grep k10temp
k10temp                16384  0
```
It seems like I didn't blacklist properly k10temp, and both modules are loaded at startup.

closed 2020-03-09 by G4E