ocerman/zenpower #28: Doesn't build against kernel 5.7-rc2

towo2099, 2020-04-25:
Hi, the module does not build against linux 5.7-rcX

```
towo@Ideapad:~/source/zenpower/zenpower$ cat /var/lib/dkms/zenpower/0.1.10/build/make.log
DKMS make.log for zenpower-0.1.10 for kernel 5.7.0-rc2-siduction-amd64 (x86_64)
Sa 25. Apr 15:36:50 CEST 2020
make[1]: Entering directory '/usr/src/linux-headers-5.7.0-rc2-siduction-amd64'
  CC [M]  /var/lib/dkms/zenpower/0.1.10/build/zenpower.o
/var/lib/dkms/zenpower/0.1.10/build/zenpower.c: In function 'zenpower_probe':
/var/lib/dkms/zenpower/0.1.10/build/zenpower.c:543:12: error: 'amd_nb_misc_ids' undeclared (first use in this function)
  543 |  for (id = amd_nb_misc_ids; id->vendor; id++) {
      |            ^~~~~~~~~~~~~~~
/var/lib/dkms/zenpower/0.1.10/build/zenpower.c:543:12: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:272: /var/lib/dkms/zenpower/0.1.10/build/zenpower.o] Error 1
make[1]: *** [Makefile:1735: /var/lib/dkms/zenpower/0.1.10/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.7.0-rc2-siduction-amd64'
make: *** [Makefile:27: modules] Error 2
```

towo2099, 2020-04-26:
After some research i've fond out, amd_nb_misc_ids is not exported anymore.
I was able to build the module with the following patch

```
--- zenpower.c.orig	2020-04-25 15:32:27.479486702 +0200
+++ zenpower.c	2020-04-26 21:33:52.937277258 +0200
@@ -32,8 +32,12 @@
 
 #include <linux/hwmon.h>
 #include <linux/module.h>
+#include <linux/init.h>
+#include <linux/export.h>
+#include <linux/spinlock.h>
 #include <linux/pci.h>
 #include <asm/amd_nb.h>
+#include <linux/version.h>
 
 MODULE_DESCRIPTION("AMD ZEN family CPU Sensors Driver");
 MODULE_AUTHOR("Ondrej Čerman");
@@ -57,6 +61,26 @@
 #define PCI_DEVICE_ID_AMD_17H_M70H_DF_F3    0x1443
 #endif
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 7, 0)
+static const struct pci_device_id amd_nb_misc_ids[] = {
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_K8_NB_MISC) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_10H_NB_MISC) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_15H_NB_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_15H_M10H_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_15H_M30H_NB_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_15H_M60H_NB_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_16H_NB_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_16H_M30H_NB_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_17H_DF_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_17H_M10H_DF_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_17H_M30H_DF_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_CNB17H_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_17H_M70H_DF_F3) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_AMD, PCI_DEVICE_ID_AMD_19H_DF_F3) },
+	{}
+};
+#endif
+
 #define F17H_M01H_REPORTED_TEMP_CTRL        0x00059800
 #define F17H_M01H_SVI                       0x0005A000
 #define F17H_M01H_SVI_TEL_PLANE0            (F17H_M01H_SVI + 0xC)
```

I don't know, if this is the right way, but now i can use the module even on kernel 5.7-rcX.

gardotd426, 2020-05-14:
It's broken completely now, because of the gcc upgrade. I'll file a separate issue.

gardotd426, 2020-05-14:
@towo2099 how did you get this patch to apply? Creating a file called `zenpower.patch` with this copied and pasted into it doesn't work. Says patch ends unexpectedly in the middle of the line.

gardotd426, 2020-05-14:
Nevermind, it seems he already merged the patch, but it still doesn't work for 5.7.0-rc5:
```
cat /var/lib/dkms/zenpower/44.ee05fcf/build/make.log
DKMS make.log for zenpower-44.ee05fcf for kernel 5.7.0-rc5-1-tkg-pds (x86_64)
Thu 14 May 2020 12:09:20 PM EDT
make: Entering directory '/usr/lib/modules/5.7.0-rc5-1-tkg-pds/build'
  CC [M]  /var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.o
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c: In function ‘zenpower_probe’:
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c:543:12: error: ‘amd_nb_misc_ids’ undeclared (first use in this function)
  543 |  for (id = amd_nb_misc_ids; id->vendor; id++) {
      |            ^~~~~~~~~~~~~~~
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c:543:12: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [scripts/Makefile.build:267: /var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.o] Error 1
make: *** [Makefile:1729: /var/lib/dkms/zenpower/44.ee05fcf/build] Error 2
make: Leaving directory '/usr/lib/modules/5.7.0-rc5-1-tkg-pds/build'
```
I have no idea how to fix that. Maybe you can give it a shot.

yuyuyak, 2020-06-03:
Hmm, I see no evidence of patch being merged.  I had same failure he had.  I used towo2099's patch (done by hand) and it's working fine for me on mainline 5.7.0.  I diff'ed the hand-patched zenpower.c with the original, it is identical to towo2099's patch.  Thanks towo2099.

yuyuyak, 2020-06-03:
Oh I see, you just need to modify the first two lines of the patch to:
--- a/zenpower.c	2020-06-02 18:38:51.120697107 -0700
+++ b/zenpower.c	2020-06-02 19:41:23.111145620 -0700
(just add the a/ and b/).  Then it works fine with patch -p1 < ***WHATEVER_NAME***.patch.  Tested.

terencode, 2020-06-03:
Or you can also just use `-p0` instead of `p1` :)

yuyuyak, 2020-06-03:
Thanks terencode, that works, learn something new everyday.

ocerman, 2020-06-06:
Hello everyone. I have pushed new version (0.1.11) that is compatible with kernel 5.7.x.

closed 2020-06-06 by ocerman