ocerman/zenpower #29: Merged 5.7-rc Patch Doesn't (or at least no longer) works

gardotd426, 2020-05-14:
So, zenpower as a whole is completely broken now on GCC 10.1.0, which was released the other day. It won't even install on LTS kernels. But I downgraded GCC to get this installed, and turns out, it doesn't work for 5.7-rc5, the current rc for 5.7. I noticed someone else had mentioned a patch, which you seem to have merged, but it doesn't work:
```
cat /var/lib/dkms/zenpower/44.ee05fcf/build/make.log
DKMS make.log for zenpower-44.ee05fcf for kernel 5.7.0-rc5-1-tkg-pds (x86_64)
Thu 14 May 2020 12:09:20 PM EDT
make: Entering directory '/usr/lib/modules/5.7.0-rc5-1-tkg-pds/build'
  CC [M]  /var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.o
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c: In function ‘zenpower_probe’:
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c:543:12: error: ‘amd_nb_misc_ids’ undeclared (first use in this function)
  543 |  for (id = amd_nb_misc_ids; id->vendor; id++) {
      |            ^~~~~~~~~~~~~~~
/var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.c:543:12: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [scripts/Makefile.build:267: /var/lib/dkms/zenpower/44.ee05fcf/build/zenpower.o] Error 1
make: *** [Makefile:1729: /var/lib/dkms/zenpower/44.ee05fcf/build] Error 2
make: Leaving directory '/usr/lib/modules/5.7.0-rc5-1-tkg-pds/build'
```

That's an error in zenpower.c, and it seems specifically to do with the new patch. 

So, in summation, zenpower is completely broken with GCC 10.1, and with 9.3 it works for other kernels, but not 5.7-rc5 because of something wrong in the patch.

abucodonosor, 2020-05-14:
The second thing you posted is a kernel -> GCC *version* mismatch nothing to do with this module at all. If you enable gcc plugins you have to compile the kernel and any other modules with the same gcc version.

Also what patch are you talking about?

The first error you posted is about a missing include file or similar, the kernel may have changed in 5.7, not a gcc issue.

gardotd426, 2020-05-14:
I wasn't suggesting that they were both gcc issues, I apologized. I've never gotten it to work with a 5.7-rc, even before the GCC update. And yes, someone pointed out that it was a version mismatch, I'm in the process of fixing that to correct the second error. 

The issue that I'm actually posting here (apologies for the confusion), the first error, and the patch I was mentioning, stemmed from #28 where someone commented that they had figured out a way to get it to build, with a patch, but it looked like some of that had already been added, but I looked at the wrong directory, where I'd already tried to apply the patch, so it looked like you'd merged it. 
[This is what I'm referring to](https://github.com/ocerman/zenpower/issues/28#issuecomment-619616487).

So basically, there's something that's changed in the kernel that breaks compatibility from 5.7, is what you're saying.

gardotd426, 2020-05-14:
I'm going to close this as a duplicate of #28, then, as it's the same issue. Can't get their patch to apply because of unexpected end of file, so I'll just wait for you to fix it or it to otherwise get sorted

closed 2020-05-14 by gardotd426

abucodonosor, 2020-05-14:
The patch seems to apply just fine.

```

crazy@nitro5:~/z$ patch -Np0  <../issue-28.patch
patching file zenpower.c

```

There a re-diffed patch but it is the same one:

https://crazy.dev.frugalware.org/issue-28.patch

gardotd426, 2020-05-14:
I don't know what the problem was, but trying the one you linked did indeed
work, I rebuilt the kernel with gcc10, applied the patch, built zenpower,
and it installed and applied correctly to 5.7-rc5 (and the others too,
obviously because that was the gcc issue). So yeah, the patch does seem to
work in at least getting it to apply, I'll reboot now and make sure it
actually works-works.

gardotd426, 2020-05-14:
I can't get it to work, at least not enough to get any debug info, the bash script to get debug info doesn't work, and i have no /sys/class/hwmon/hwmon*/debug_data. 

The problem seems to be k10temp, it won't blacklist no matter what I do. `sensors-detect` still detects it as the module for cpu temps, I've regenerated the initramfs, k10temp is in `/etc/modprobe.d/blacklist.conf`, and no matter what I do, k10temp and zenpower both get loaded at boot. It's driving me crazy, but that's obviously not a zenpower issue.