ocerman/zenpower #30: zenpower data

jave808, 2020-05-18:
Ryzen 7 2700X
Gigabyte B450M-DS3H Rev 1.0
Linux 5.3.0-51

KERN_SUP: 1
NODE0; CPU0; N/CPU: 1
0005a008 = 00000002
0005a00c = 0140008f
0005a010 = 0176001d
0005a014 = 00000000
000598bc = 0fff00ff
0005994c = 00000000
00059954 = 00000000
00059958 = 00000000
0005995c = 00000000
00059960 = 00000000
00059964 = 08400001
00059968 = 00008a45
0005996c = 0000004b
00059970 = c0800005

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.15 V  
SVI2_SoC:     +0.81 V  
Tdie:         +58.0°C  (high = +95.0°C)
Tctl:         +68.0°C  
SVI2_P_Core: 172.09 W  
SVI2_P_SoC:    8.51 W  
SVI2_C_Core: +150.69 A  
SVI2_C_SoC:  +10.46 A

birdie-github, 2020-05-21:
The zenpower driver doesn't provide this data. File a bug report against Zen Monitor instead.

ocerman, 2020-06-06:
Hello, thx for your report. This issue is in zenmonitor app. It may be related to ocerman/zenmonitor#20

closed 2020-06-06 by ocerman