ocerman/zenpower #32: How to enable on Manjaro/Arch

bushblade, 2020-06-24:
Hi, I've installed this via the AUR on Manjaro, but running
`sudo modprobe zenpower` as per the README
I get
`modprobe: FATAL: Module zenpower not found in directory /lib/modules/5.6.16-1-MANJARO`

Benzhaomin, 2020-06-25:
Hi, which package did you install from the AUR?

https://aur.archlinux.org/packages/?O=0&K=zenpower

If you're using a dkms package you don't need to modprobe the module.

bushblade, 2020-06-25:
Hi thanks for the reply.
I use [zenper-dkms](https://aur.archlinux.org/packages/zenpower-dkms) from the AUR.
So how do I use it then?

Benzhaomin, 2020-06-26:
I suppose you had just upgraded your kernel and zenpower was built for another version. Simply rebooting or reinstalling after having rebooted should be enough.

bushblade, 2020-06-26:
rebooting was the first thing i tried

jmg87, 2020-06-27:
what does this get you:

 ╱  ~                                                                   at 23:39:55  ─╮
❯ dkms status                                                                              ─╯
acpi_call, 1.1.0, 5.6.0-1-amd-staging-drm-next-git-01003-gcd5dd023c24f, x86_64: installed
acpi_call, 1.1.0, 5.7.6-arch1-1, x86_64: installed (WARNING! Missing some built modules!)
exfat, 206.297a573, 5.6.0-1-amd-staging-drm-next-git-01003-gcd5dd023c24f, x86_64: installed
r8168, 8.048.03, 5.6.0-1-amd-staging-drm-next-git-01003-gcd5dd023c24f, x86_64: installed
r8168, 8.048.03, 5.7.6-arch1-1, x86_64: installed (WARNING! Missing some built modules!)
r8169_aspm, v4.15.3: added
vboxhost, 6.1.10_OSE, 5.6.0-1-amd-staging-drm-next-git-01003-gcd5dd023c24f, x86_64: installed
vboxhost, 6.1.10_OSE, 5.7.6-arch1-1, x86_64: installed (WARNING! Missing some built modules!) (WARNING! Missing some built modules!) (WARNING! Missing some built modules!)
zenpower, 0.1.12, 5.6.0-1-amd-staging-drm-next-git-01003-gcd5dd023c24f, x86_64: installed <- good
zenpower, 0.1.12, 5.7.6-arch1-1, x86_64: installed (WARNING! Missing some built modules!) < - wont work without rebuild

You sure you have the headers for your current version of kernel? In theory dkms handles most of the hassle for you, but with no headers, no module.

Once you install them, run 
> dkms autoinstall 

and should be ok

xythrez, 2020-06-28:
@bushblade AUR package maintainer here, do you have the corresponding linux-headers (or linux56-headers for Manjaro) installed? It's technically a dependency but I can't put it in due to [DKMS package guidelines](https://wiki.archlinux.org/index.php/DKMS_package_guidelines#Dependencies). Let me know if installing it fixes your problem.

bushblade, 2020-06-28:
Hi @xythrez thanks for taking the time to answer.
Yeah pretty sure I do...?
[image]
@jmg87 I don't get any output from dkms status

xythrez, 2020-06-28:
@bushblade I believe that the linux56-rt-headers is for the linux-rt kernel, there should be a separate headers package called "linux56-headers" for the kernel you are using. The easiest way to confirm would be installing any kernel related packages via pacman. If `pacman -Syu linux56-headers` shows `Installing DKMS Modules` with no errors, then the zenpower should at least be installed. Otherwise, the module may have failed to compile, which would explain why it is missing.

bushblade, 2020-06-28:
Ah that did indeed install the module.

bushblade, 2020-06-28:
Ah that did indeed install the module.
And all is working now, I can read my CPU temps just fine.
Thanks so much for the help.

closed 2020-06-28 by bushblade