ocerman/zenpower #33: Interferes with NBFC module on ASUS ROG GL702ZC

thyTwilightGoth, 2020-07-01:
it disables nbfc's fan control, which is the only way to control the fans on the ASUS ROG GL702ZC on Linux.

abucodonosor, 2020-12-25:
Hmm? How so?

NBFC is operating on the EC of the Laptop poking the custom fan control.

However, is zenpower breaks it so will k10temp. But I cannot confirm that, I don't have an ASUS Laptop but an ACER one,
both are using the same interface.

Yato202010, 2021-03-14:
this also happen on acer laptop here is a issue i have open on nbfc github
https://github.com/hirschmann/nbfc/issues/1096

batmalin, 2021-03-19:
This is because it uses coretemp and k10temp to monitor the CPU temperature. You have to uninstall it/remove it by deleteing NBFC folder from /opt and /home folders to download the source again and to go to Core/Plugins/StagWare.Plugins.FSTemperatureMonitor/FSTemperatureMonitor.cs , open it and edit line 28 to include "zenpower" , save and then compile it again and it  works.