ocerman/zenpower #36: Does Zenower support the PCI ids for Rome Epyc Zen 2 cpus?

KeithMyers, 2020-08-19:
Does Zenpower support the PCI ids for Rome Epyc Zen 2 cpus?

abucodonosor, 2020-12-25:
Yes, it does.

KeithMyers, 2020-12-25:
Yes, I figured it out after I upgraded to a Epyc 7402P from a Threadripper Zen+ 2920X cpu.
I posted my Epyc 7402P Zenpower debug script outputs in the script issue thread.
But Zen Monitor does not work on the higher core count cpus according to my friend with 32 and 64 core parts.

abucodonosor, 2020-12-25:
@KeithMyers 

I don't have a Rome box, but what exactly isn't working?

KeithMyers, 2020-12-25:
Zen Monitor won't run as root on 48 core Epyc 7642 for example when you want to get the power readings.
No response from ocerman at all on the issue.
[https://github.com/ocerman/zenmonitor/issues/34](url)

abucodonosor, 2020-12-25:
@KeithMyers 

I don't use Ubuntu, but what means 'run as root' ?.. Have a GUI and login as root, use sudo, use pkexec, use xdg-su, gksu,
kdesu .. etc ( as a note xdg-su is something no one should use ).

@ocerman didn't response on any issue in a long time, I hope he is doing well.

KeithMyers, 2020-12-25:
When you install Zen Monitor with the policy kit, you end up with two Zen Monitor icons created in Apps.  The normal user one just displays the voltages, clocks and temps from ZenPower.

But the one created to be run as root user displays all of the ZenPower parameters including power readings from the RAPL interface.

What the comment was saying what that the application will not run as root user on the higher core count parts. It just hangs.  Speculation was the app did not anticipate the high core count parts and does not have the necessary layout size to display all the cores  and extra readouts and it hangs.

KeithMyers, 2024-05-01:
Just wanted to update this issue.  Somewhere along the way of kernel updates.  Zenmonitor3 runs fine and displays all the cores of my 64 core Epycs.  Both Zen2 and Zen3.
This issue should be closed as complete.

closed 2024-05-01 by KeithMyers