ocerman/zenpower #42: 5950x do not appear with zenpower

aenertia, 2021-09-15:
Running 5.14 Kernel on B550M and fedora. 

Module loads but sensors does not find k10temp interfaces to report.

valerii540, 2021-12-10:
The same problem with Ryzen 9 5900HX

OS: Manjaro 21.2.0 Qonos
Kernel: 5.15.2-2-MANJARO

birdie-github, 2021-12-19:
This driver hasn't been updated in months and it's not clear whether its developer will resume work on it.

Please use https://github.com/Ta180m/zenpower3 in the meanwhile.

epifanio, 2021-12-19:
> This driver hasn't been updated in months and it's not clear whether its developer will resume work on it.
> 
> Please use https://github.com/Ta180m/zenpower3 in the meanwhile.

indeed the new version/drivers are working great! thanks!

[image]