ocerman/zenpower #44: cpu power

Dany-Wilde, 2022-12-28:
Hello to all.

happy new year to come .

here it is : I am under debian sid kernel 6.0.0-6-amd64

i noticed (at least, it seems to me) that the display of the cpu power (ryzen 9 3900x, asus prime x570 pro) in lm-sensors does not work anymore. 
the voltages have also disappeared.
 only the temperatures and the speed of the cpu fan remain.

Do you have any explanation for this phenomenon?
and solutions ?

in my games, with mangohud, I would have liked to display also the cpu power.
but this data is constantly displayed at zero.

waiting to read you ...
friendly, dany .

p.s.: on request, I can give you more information, of course.

Translated with www.DeepL.com/Translator (free version)

Dany-Wilde, 2022-12-28:
alain@sid:~$ sensors 
jc42-i2c-0-1b
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +41.0°C  (low  =  +0.0°C)                  ALARM (HIGH, CRIT)
                       (high =  +0.0°C, hyst =  +0.0°C)
                       (crit =  +0.0°C, hyst =  +0.0°C)

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:      775.00 mV 
fan1:        1471 RPM  (min =    0 RPM, max = 3400 RPM)
edge:         +36.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +37.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +38.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
PPT:          33.00 W  (cap = 289.00 W)

nvme-pci-0100
Adapter: PCI adapter
Composite:    +37.9°C  (low  = -60.1°C, high = +89.8°C)
                       (crit = +94.8°C)

asusec-isa-0000
Adapter: ISA adapter
Chipset:     1438 RPM
Chipset:      +58.0°C  
CPU:          +40.0°C  
Motherboard:  +29.0°C  
T_Sensor:     -40.0°C  
VRM:           +0.0°C  

jc42-i2c-0-1a
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +39.8°C  (low  =  +0.0°C)                  ALARM (HIGH, CRIT)
                       (high =  +0.0°C, hyst =  +0.0°C)
                       (crit =  +0.0°C, hyst =  +0.0°C)

nvme-pci-0400
Adapter: PCI adapter
Composite:    +36.9°C  (low  =  -0.1°C, high = +84.8°C)
                       (crit = +94.8°C)
Sensor 1:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 8:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)

alain@sid:~$ uname -a 
Linux sid 6.0.0-6-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.12-1 (2022-12-09) x86_64 GNU/Linux

alain@sid:~$ lscpu | grep "modèle"
Nom de modèle :                         AMD Ryzen 9 3900X 12-Core Processor

alain@sid:~$ sudo dmesg | grep DMI:
[    0.000000] DMI: System manufacturer System Product Name/PRIME X570-PRO, BIOS 4204 02/24/2022

in french : 
[https://debian-facile.org/viewtopic.php?id=32809](url)

EUA, 2023-02-04:
I use "zenpower" kernel module to track watts..

      zenpower-pci-00c3
      Adapter: PCI adapter
      SVI2_Core:     1.30 V  
      SVI2_SoC:      1.09 V  
      Tdie:         +71.6°C  (high = +95.0°C)
      Tctl:         +71.6°C  
      SVI2_P_Core:  68.34 W  
      SVI2_P_SoC:    4.80 W  
      SVI2_C_Core:  52.70 A  
      SVI2_C_SoC:    4.41 A  
      
      amdgpu-pci-0a00
      Adapter: PCI adapter
      vddgfx:        1.31 V  
      vddnb:         1.09 V  
      edge:         +53.0°C  
      PPT:          76.00 W  

unfortunatelly, freaon doesn't show power. Just temperatures, fan speeds and voltages... :-1: