ocerman/zenpower #47: Is this project dead?

nforced, 2023-03-12:
Considering the issues, commits, Zen 4 support and overall activity on the project is seems abandoned.
Is the anyone maintaining this repository or any plans to get it updated?

KeithMyers, 2024-05-01:
No support for Zen4.  Assume someone would have to figure out the ID's and families etc.
But you can still use Zenmonitor3 on Zen4 because it can read from the RAPL registers for power and clocks.
You just don't get any of the SVI2_ registers like voltages or currents.

nforced, 2024-05-02:
@KeithMyers probably but hey newer kernels have this built in! So we no longer need this to read temp etc :wink:

KeithMyers, 2024-05-02:
The temp reading is still iffy with the SIO chipset drivers.  k10temp is the only dependable way.  But you need the chipset drivers to read fans and voltages.  Fans for the most part are OK, but the voltages are out to lunch.  You just have to do some guesswork in what the named inputs might be related to and then kludge some sensors3.conf values to sort of put inputs like Vcpu somewhere in a reasonable range.  All made very difficult because the SIO chipset makers won't release the technical docs on their chips.

nforced, 2024-05-02:
Could be I'm now on b650 chipset with 7950x but to be honest no longer care about voltages and temps with arctic 240 liquid cooling set to low and bios setting to auto it just silent and chill

KeithMyers, 2024-05-02:
If you never really stress your cpu or system I guess that works.

I OTOH, run all my hosts flat out 24/7 doing distributed computing.  Even with custom liquid cooling on a gpus and cpus with multiple rads, I am always pushing 95° C. on the Zen 4 hosts.  Same for the 5950X.  Only the Epyc servers I don't have to really worry about temps since they run in the high 40's, low 50's all the time. I like to see where the temps are at all times on all hosts.  Most of the gpus are EVGA Hybrids which are known to lose pumps and even with the custom rads, you need to run your fans at full speed, so any loss of fan speed or fan failures will push the temps to shutdown limits and also error out all the work I run.