ocerman/zenpower #50: Build broken against Linux 6.14

Dietr1ch, 2025-03-26:
When building the module for 6.14 on NixOS I'm getting an error about `amd_pci_dev_to_node_id` being undefined,

    > zenpower.c: In function 'zenpower_probe':
    > zenpower.c:587:41: error: implicit declaration of function 'amd_pci_dev_to_node_id' []
    >   587 |                         data->node_id = amd_pci_dev_to_node_id(pdev);
    >       |                                         ^~~~~~~~~~~~~~~~~~~~~~

Folks from CoreFreq have this patch that seems relevant.
  - https://github.com/cyring/CoreFreq/commit/cb9fc92098b90ef6992e065d0ceac76ea0ff6fe1

I didn't took the time to dig into why this change is needed on newer kernels, but it seems the function isn't there anymore because there's a more general alternative to get the same data

stele95, 2025-04-10:
Use this one, it has the needed patch
https://github.com/AliEmreSenel/zenpower3

dasmikko, 2025-05-15:
> Use this one, it has the needed patch https://github.com/AliEmreSenel/zenpower3

omg thank you! This has been bothering me for months.